MycroftAI/enclosure-picroft #13: automated build

jtheriault, 2017-03-18:
I have [a fork](https://github.com/jtheriault/enclosure-picroft/tree/travis_build) that adds a build process that overlays the Picroft code onto an existing SD card image. At the moment I use a past Picroft build, but that's obviously not ideal. What would be ideal would be to overlay it on a Mycroft standalone image.

2 questions:
1. Before I go about building it myself, is there already an available location of SD card images of Mycroft standalone?
1. Would this be interesting as a pull request?

kfezer, 2017-03-24:
I don't think I quite understand what you mean. Picroft is really the only standalone Mycroft SD Card image as it's just Mycroft pre-installed on a Rasbian Lite image.

jtheriault, 2017-03-25:
Sorry. I was thinking that it was possible that, since there's more to a Picroft image that just what is in this repo, that it might build upon another image. I was guessing (and neglected to be specific) that the image used for the Pi in the Mark 1 might supply just such a base.

Also I apologize for the confusing use of "standalone." I had just been devising plans of how to use the standalone.sh script on bootstrap.mycroft.ai to create my image and I guess the name creeped into my word choices.

Should I take it that there isn't a ready image to use as a base?

penrods, 2017-12-06:
Sorry, this question got dropped and never answered.

The Picroft image is prebuilt for a plain-old Raspberry Pi 3.  You can get the latest at https://mycroft.ai/to/picroft-image

I hope this helps!

closed 2017-12-06 by penrods