MycroftAI/enclosure-picroft #22: how to change the wake word in picroft

raja434, 2017-05-28:
hi how can i change the wake word in picroft

forslund, 2017-05-28:
Hi, to change wake word go to https://home.mycroft.ai and then to settings -> advanced
and enter the wakeword and corresponding phonemes under _Listener_*.

Phonemes for words can be looked up using http://www.speech.cs.cmu.edu/cgi-bin/cmudict

penrods, 2017-12-06:
Answered, closing.

closed 2017-12-06 by penrods

kmccurley, 2018-04-14:
I believe that this answer only applies to the older PocketSphinx system, but not precise?

penrods, 2018-04-15:
Precise will only be used if there is a model for the given wakeword.  Currently the only published model is "Hey Mycroft" (although you can build your own if you want to try that).

When there is no model for the wakeword, mycroft-core will fall back to using PocketSphinx.