MycroftAI/enclosure-picroft #24: Default password is now 'mycroft'

gboudreau, 2017-06-12:
The default password is `raspberry`, not `mycroft` as stated in the documentation.

When using `sudo -i`, I see this, which is also shown on boot:

```
SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.
```

gboudreau, 2017-06-12:
Might be related: saw error messages about invalid certificates, when apt-get was trying to upgrade packages from repo.mycroft.ai.
Might be caused by invalid date/time on the device ..?
Also seeing 401 errors when trying to use `say` command, even after I linked the device to my account on home.mycroft.ai

mikeecuffe, 2017-06-24:
thanks gboudreau! I couldn't find this info anywhere else.
I symlink'd /etc/localtime and the logs are correct but say: "what time is it?" still reports wrong time zone. 
I also had the invalid cert errors and will monitor to see if this resolved..

still. its the first week and baby steps it is

forslund, 2017-06-24:
@mikeecuffe, have you set the device location at home.mycroft.ai (devices->[your device]->location

Mycroft uses that rather than system settings to handle timezone (as far as I know)

mikeecuffe, 2017-06-29:
@forslund thank for advice. I took your suggestion and set the time zone @ home.mycroft.ai but still receive incorrect time.

mikeecuffe, 2017-06-29:
The invalid cert errors cleared up once the date was correct locally.
 My timezones issue is unrelated and I won't hijack this thread any further.

penrods, 2017-09-24:
Which documentation says the password is "mycroft"?  That is only the case for the Mark 1, we intentionally left Picroft at pi/raspberry just like a normal Raspbian install.  If there is doc that is wrong or unclear, I'd like to clear that up.

gboudreau, 2017-09-24:
https://github.com/MycroftAI/enclosure-picroft/wiki/HOWTO_audio_output

penrods, 2017-09-24:
Ah, that was a user contribution on the wiki.  I can correct it -- or you are welcome to!  The wiki is freely editable by anyone.

jharbott, 2018-01-13:
This also shows the wrong password: https://mycroft.ai/documentation/picroft/#connecting-to-picroft-via-ssh

penrods, 2018-03-15:
I've been convinced that 'mycroft' _is_ the actual correct password.  :)

The lateset images all use this.

closed 2018-03-15 by penrods

benjaffe, 2020-09-05:
For clarification, username is `pi`, password is `mycroft`, at least in my experience.

TupaNegreiros, 2020-12-29:
Password doesn't seems to be `mycroft`, nor `raspberry`, nor `pi`. I had to change with `sudo passwd pi`.