MycroftAI/enclosure-picroft #25: apt-get: server certificate verification failed

gboudreau, 2017-06-12:
Seen during boot of newly flashed picroft SD card:

```
Checking for updates to Mycroft...
Ign http://repo.mycroft.ai debian InRelease
Ign http://repo.mycroft.ai debian Release.gpg
Ign http://repo.mycroft.ai debian Release
Err http://repo.mycroft.ai debian/main armhf Packages

Err http://repo.mycroft.ai debian/main armhf Packages

Err http://repo.mycroft.ai debian/main armhf Packages

Err http://repo.mycroft.ai debian/main armhf Packages

Err http://repo.mycroft.ai debian/main armhf Packages
  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none
Ign http://repo.mycroft.ai debian/main Translation-en_GB
Ign http://repo.mycroft.ai debian/main Translation-en
W: Failed to fetch http://repo.mycroft.ai/repos/apt/debian/dists/debian/mai/binary-armhf/Packages  server certificate verification failed. CAfile: /etc/ssl/certs/ca-certificates.crt CRLfile: none

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

Trying `sudo apt-get update && sudo apt-get update` from a SSH session works without problem.

gboudreau, 2017-06-12:
Looks like updating apt-get packages is causing an infinite loop of 401 errors as seen in view_log...
Happened after I registered once, then I re-flashed, and tried to do apt-get update/upgrade before registering, and same thing happened.

BenjaminHCCarr, 2017-06-25:
We've tracked this down.... the GPG key went sideways (really it timed out but as the warning is in UNIX time since EPOCH it's kinda hard to keep track of)....
The solution is to run:
`sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys F3B1AA8B`

This will add the newly signed ket to the `apt` keyring.

penrods, 2017-12-06:
Please update to the 0.9 image.  https://mycroft.ai/to/picroft-image

This is resolved there.

closed 2017-12-06 by penrods