MycroftAI/enclosure-picroft #28: hello devs i got an error

jrsarath, 2017-06-24:
im getting this 
 loop = RecognizerLoop()
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.1-py2.7.egg/mycroft/client/speech/listener.py", line 184, in __init__
    self.mycroft_recognizer = self.create_mycroft_recognizer(rate, lang)
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.1-py2.7.egg/mycroft/client/speech/listener.py", line 194, in create_mycroft_recognizer
    return LocalRecognizer(wake_word, phonemes, threshold, rate, lang)
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.1-py2.7.egg/mycroft/client/speech/local_recognizer.py", line 40, in __init__
    self.decoder = Decoder(self.create_config(dict_name))
  File "/usr/local/lib/python2.7/site-packages/pocketsphinx/pocketsphinx.py", line 271, in __init__
    this = _pocketsphinx.new_Decoder(*args)
RuntimeError: new_Decoder returned -1

penrods, 2017-12-06:
Still an issue with the 0.9 image?  https://mycroft.ai/to/picroft-image

jrsarath, 2017-12-08:
didn't tried yet, will try and update

penrods, 2018-03-15:
Assuming fixed since I haven't heard anything more

closed 2018-03-15 by penrods