MycroftAI/enclosure-picroft #30: test_microphone utility is not working

stevedee, 2017-06-27:
PiCroft test_microphone or mycroft-audio-test does not work on my Pi-3, although mycroft works fine when in use.

The test_microphone utility would be very handy for testing the speech quality of a range of microphones, so I can use the best that I have.

Please let me know if you need more info.

Screen dump:-
$ test_microphone
Shutting down mycroft...
Starting a 5 second test recording...
2017-06-27 10:34:15,154 - mycroft.configuration - INFO - Loading configuration: /usr/local/lib/python2.7/site-packages/mycroft_core-0.8.16-py2.7.egg/mycroft/configuration/mycroft.conf
2017-06-27 10:34:15,156 - mycroft.configuration - DEBUG - Configuration '/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.16-py2.7.egg/mycroft/configuration/mycroft.conf' loaded
2017-06-27 10:34:15,157 - mycroft.configuration - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' loaded
2017-06-27 10:34:15,157 - mycroft.configuration - INFO - Loading configuration: mycroft.ai
2017-06-27 10:34:15,631 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
2017-06-27 10:34:16,071 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/device//setting HTTP/1.1" 401 38
2017-06-27 10:34:16,088 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
2017-06-27 10:34:16,518 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/auth/token HTTP/1.1" 401 38
2017-06-27 10:34:16,539 - mycroft.configuration - WARNING - Failed to fetch remote configuration: HTTPError(u'The supplied authentication is invalid',)
2017-06-27 10:34:16,540 - mycroft.configuration - INFO - Loading configuration: /etc/mycroft/mycroft.conf
2017-06-27 10:34:16,542 - mycroft.configuration - DEBUG - Configuration '/etc/mycroft/mycroft.conf' loaded
2017-06-27 10:34:16,543 - mycroft.configuration - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' loaded
2017-06-27 10:34:16,544 - mycroft.configuration - INFO - Loading configuration: /home/pi/.mycroft/mycroft.conf
2017-06-27 10:34:16,546 - mycroft.configuration - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' loaded
2017-06-27 10:34:16,547 - mycroft.configuration - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' loaded
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1286:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4259:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4738:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Playing WAVE '/tmp/test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
Playing back the recording...
Restarting mycroft...

forslund, 2017-06-27:
Hi,

the script seems to be running to completion. I guess your issue is that no sound gets recorded while the test is running?

stevedee, 2017-06-28:
Yes, I'm sorry I should have made that clear. During test_microphone playback I just hear background noise. I assume this utility is recording the wrong channel/device for my Adafruit USB audio adapter, although audio is fine when using Mycroft (i.e. the system responds to my commands and I hear Mycroft reply).

forslund, 2017-06-28:
Ok, I see. Do you have multiple mics?

The test tool should use the same channel/device as the normal mycroft settings. Which user are you running the tests under? mycroft or pi?

stevedee, 2017-06-28:
No, I only have one mic connected: https://community.mycroft.ai/t/easiest-mic-speaker-option-for-raspberry-pi-3/1246/31

I'm running as user "pi" which I guess is the default user on Picroft.

penrods, 2018-03-15:
This works in the latest image.  Also, the mic meter is now built-in for the CLI.

closed 2018-03-15 by penrods

khklatt, 2018-03-31:
I'm using Picroft dated 2018-03-14?  Using alsa, I can see the microphone, and can adjust the gain -- Script runs to completion after setting the gain, but no sound..   Alsa mixer shows:
| Card: bcm2835 ALSA                                   F1:  Help               |
| Chip: Broadcom Mixer                                  F2:  System information |
| View: F3:[Playback] F4: Capture  F5: All      F6:  Select sound card  |
| Item: PCM [dB gain: -3.48]                           Esc: Exit            

Are there OS commands I can use from the ssh session to check if a valid microphone is found?

khklatt, 2018-04-01:
Update -- After setting the gain, picroft now seems to recognize my voice.  Yeah!