MycroftAI/enclosure-picroft #31: Default Arduino Faceplate port blocks bluetooth on Pi 3

RotatingFans, 2017-07-06:
The Pi 3 uses /dev/ttyAMA0 to communicate with the bluetooth module through UART. Changing the config file to change the port to /dev/ttyAMA1 allows bluetooth to function

Landrash, 2017-07-07:
The raspberry pi 3 only has one serial port and it shares that with the Bluetooth module.
If you haven't physically changed anything I thinks it's a tad strange that you can keep the Mycroft unit fully functioning with both Bluetooth and the Faceplate working?

RotatingFans, 2017-07-07:
According to https://raspberrypi.stackexchange.com/questions/45570/how-do-i-make-serial-work-on-the-raspberry-pi3 the pi 3 still has 2 UARTS, 1 full UART default used for Bluetooth and 1 microUART mapped to pins, and newer firmware will automatically map the non bluetooth one to /dev/serial0. Disabling Bluetooth and remapping pins would allow for full UART on /dev/ttyAMA0 in comparison to microuart available while bluetooth is in use.

It also may not need to be a default setting, just a note on the README to help anyone who can't get bluetooth to work with picroft as the problem is not inherently obvious.

RotatingFans, 2017-07-07:
Also, I see the faceplate mentioned, but what exactly is the faceplate?

Landrash, 2017-07-07:
I wasn't aware that the pi 3 had 2 UARTS. Til.
The Faceplate of the Mycroft mark 1 unit is an Arduino and some electronics that give the Mycroft unit it's "face and eyes" It's connected to the pin headers on the pi :)

Celti, 2018-05-24:
Bluetooth works fine on the Mark 1 (even for Bluetooth audio!) with the following configuration changes in `/boot/config.txt`

```
# Disable Bluetooth, it interferes with serial port
#dtoverlay=pi3-disable-bt
dtoverlay=pi3-miniuart-bt
core_freq=250
```

Make sure that `hciuart.service` and `bluetooth.service` are enabled — for me, they were by default and just unable to work — and everything should "Just Work™". The `hciuart` service automatically looks at `/dev/serial1`, which is always linked to the UART the bluetooth is using — in this case, `/dev/ttyS0` — while `/dev/serial0`/`/dev/ttyAMA0` remains free for the faceplate.

RotatingFans, 2018-05-24:
Currently that is the case. However, in pror versions of the raspberry pi firmware/software/hardware(I don't know which), it was the way I stated earlier. We shouls keep both options in the documentation as has been done to ensure it woeks across all versions.

penrods, 2019-07-12:
Picroft no longer uses any of the Mark 1 packaging, so you should be free to use Bluetooth.

closed 2019-07-12 by penrods