MycroftAI/enclosure-picroft #32: Tested on Pi Zero?

krishnakanthpps, 2017-07-20:
Hi. 

Has this been tested on Pi Zero W. Also, if it is connected to a HDMI device which has both display and audio, would it send audio to the HDMI connected device? or I use a sound card would it work?

This is specifically for the Pi Zero W version. 

Or is the processing power of the Pi Zero not good enough for the Picroft to work well?

forslund, 2017-07-25:
I don't think anyone has tested. The processing power of the Zero looks a bit on the poor side for running Mycroft in a good way at least at the moment.

penrods, 2017-09-24:
You will not be able to run well on a PiZero if at all.  Mycroft requires a Pi3 to run well, even on a Pi2 the experience is very laggy.

penrods, 2017-12-06:
Closing as an issue.

closed 2017-12-06 by penrods