MycroftAI/enclosure-picroft #34: Changes in home.mycroft.ai aren't applied to the PiCroft and picroft doens't recognize anything

dvdbot, 2017-08-31:
I've changed the following in the online panel:
Settings > Advanced > Text-to-Speech Engine > Voice: Female (STL)
Settings > Advanced > Listener > WAKE WORD: hey sarah
Settings > Advanced > Listener > PHONEMES: HH EY . S EH R AH .
Rebooted the Pi but they changes aren't applied to the Pi...

When asking the Pi something, like: "Hey Mycroft, what time is it?"
I get a respond that he didn't understand what I said, in the log there's an error: "Failed to find intent"
If I check the log I also see MyCroft understood what I said: Utterance [u"what time is it"]
But why is there a u withing the [ ], maybe this causes the problems?

forslund, 2017-08-31:
Hi,

we're currently having some issues with the servers and the settings made at home.mycroft.ai are very unreliable. We're working hard at getting things up and stable again.

Regarding the failed to find intent error it sounds like the skills haven't been downloaded correctly. Can you try to run msm default from the command line.

dvdbot, 2017-08-31:
Getting this back when running `msm default`:
`#######  Mycroft Skill Manager #######
msm: Mycroft Skill Manager
     Copyright (c) 2017 Mycroft AI, Inc.  All rights reserved.
usage: msm install <repository>
     Installs the given Skill into the /opt/mycroft/skills directory
     where <repository> is the address of the Skill in Github.
example: msm install https://github.com/ethanaward/demo_skill.git
`

dvdbot, 2017-08-31:
time command works, the following skills are installed, where can I see the progress?
skill-alarm  skill-audio-record  skill-configuration  skill-date-time  skill-desktop-launcher  skill-ip

dvdbot, 2017-08-31:
MyCroft is slowly installing the apps, thanks for your help, this can be closed :)

closed 2017-08-31 by dvdbot