MycroftAI/enclosure-picroft #36: 4GB card too small

NicoVermeir, 2017-09-11:
When trying to flash the 0.8 image to a 4GB card I get an error that the available space is too small. According to the requirements it should be enough. So either the requirements, or Etcher is wrong :)

KathyReid, 2017-09-24:
Thanks @NicoVermeir for raising this issue. 
Are you able to give us the terminal output so that we can verify, and update the requirements?
Many thanks for reporting this - it's super helpful. 
Kind regards, 
Kathy

penrods, 2017-09-24:
@NicoVermeir, if you have another 4GB card can you give that a shot?  Often SD cards have less than the advertised size actually available due to bad sectors on the card.  So an image taken of one 4GB card might be just slightly too large to fit on another 4GB card.  Additionally, I don't believe I was using "PiShrink" when I produced that image so it is full-sized for a 4GB card at 3,907,584 bytes.

If you have an 8 GB card I would recommend using it for a variety of reasons (including the above).  Once you start adding your own code and skills, you don't want to run out of space and have to start all over.

penrods, 2017-12-06:
Updated the readme.   Closing.

closed 2017-12-06 by penrods