MycroftAI/enclosure-picroft #38: python error in the listener.py

tjoen, 2017-09-26:
After updating to the the 0.8.22 version, I get a python error in the listener.py.

    load_entry_point('mycroft-core==0.8.22', 'console_scripts', 'mycroft-speech-client')()
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.22-py2.7.egg/mycroft/client/speech/main.py", line 140, in main
    loop = RecognizerLoop()
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.22-py2.7.egg/mycroft/client/speech/listener.py", line 193, in __init__
    self._load_config()
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.22-py2.7.egg/mycroft/client/speech/listener.py", line 211, in _load_config
    self.wakeword_recognizer = self.create_wake_word_recognizer()
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.22-py2.7.egg/mycroft/client/speech/listener.py", line 227, in create_wake_word_recognizer
    config[word]["phonemes"] = phonemes
KeyError: u'hey pedro'

0.8.21 seemed to work perfectly? 
I tried just about everything: 
- reinstalling the image
- dist upgrade to stretch, 
- installing the dev version
- rebuilding python from source, manually installing pip.

Is nobody else having this issue?

tjoen, 2017-09-26:
It actually seems to be related with:
https://github.com/MycroftAI/mycroft-core/issues/1105

But the suggested fix does not work.

forslund, 2017-09-26:
Hi,

for me it works but we're still trying it out. If you can try the PR branch or provide any extra debug info I would be grateful.

There's also a workaround for the issue by editing the ~/.mycroft/mycroft.conf (for the user running mycroft)

tjoen, 2017-09-27:
Sure, so where can I find this pr branch?

The suggested workaround of adding ~/.mycroft/mycroft.conf
does not seem to pickup my custom wakeword "hey pedro".

I also tried it in /home/pi/.mycroft/mycroft.conf because this one seems to be loaded at the end.

mycroft.conf:
{
  "listener": {
    "wake_word": "hey pedro",
    "phonemes": false,
    "threshold": false
  },
  "hotwords": {
    "hey pedro": {
        "module": "pocketsphinx",
        "phonemes": "HH EY . P EY D R OW",
        "threshold": 1e-90
      }
  }
}

When changing the name back to hey mycroft in the online webinterface it does work, as 'hey mycroft'.
It just does not pickup 'hey pedro'.

forslund, 2017-09-27:
This works for me on two different Mycroft instances using the dev branch of mycroft-core:

```json
{
        "listener": {
            "wake_word_upload": {
                "enable": false
            },
            "wake_word": "hey pedro",
            "phonemes": false,
            "threshold": false
        },

        "hotwords": {
            "hey pedro": {
                "module": "pocketsphinx",
                "phonemes": "HH EY . P EY D R OW",
                "threshold": 1e-90
            }
        }
}
```

I've copied the branch from my repository to the official mycroft-core repository so you should be able to pull down the branch `bugfix/issue-1105`

Thank you for taking the time to test.

penrods, 2017-09-28:
I merged the PR into the mycroft-core 'dev' branch just now, if you'd like to just try from there.

tjoen, 2017-09-29:
Just updated from my master branch on the pi to the dev branch. The error does not occur anymore, but the wakeup word is also not recognized. Could be my additional config maybe?

If I reburn a fresh pi dev image, I am now no longer able to register to home.mycroft.ai. It just does not pick up my registration code. 

I also tried the standard pi image, that one still does connect via home,mycroft.ai. Upgrading to the dev version by adjusting the sources.list. Now on mycroft-core (1506671182).

Changing the wakeword on home.mycroft.ai now also works, but after that, it still does not pickup the changed wakeword. No error though. Also the default wakeword 'hey mycroft' is not working anymore,

`11:38:48.687 - mycroft.client.speech.mic:listen:490 - DEBUG - Waiting for wake w                                                      ord...
11:38:48.695 - mycroft.client.speech.listener:create_wake_word_recognizer:218 -                                                       INFO - creating wake word engine
11:38:48.698 - mycroft.client.speech.hotword_factory:create_hotword:139 - INFO -                                                       creating hey pedro
11:38:48.701 - mycroft.client.speech.hotword_factory:__init__:58 - WARNING - Non                                                      e module does not match with Hotword class pocketsphinx
11:38:48.840 - mycroft.client.speech.listener:create_wakeup_recognizer:235 - INF                                                      O - creating stand up word engine
11:38:48.842 - mycroft.client.speech.hotword_factory:create_hotword:139 - INFO -                                                       creating wake up
11:38:49.986 - mycroft.client.speech.listener:run:307 - DEBUG - Config has chang                                                      ed, reloading...
11:38:50.931 - mycroft.client.speech.mic:listen:490 - DEBUG - Waiting for wake w                                                      ord...
11:38:50.938 - mycroft.client.speech.listener:create_wake_word_recognizer:218 -                                                       INFO - creating wake word engine
11:38:50.941 - mycroft.client.speech.hotword_factory:create_hotword:139 - INFO -                                                       creating hey pedro
11:38:50.943 - mycroft.client.speech.hotword_factory:__init__:58 - WARNING - Non                                                      e module does not match with Hotword class pocketsphinx
11:38:51.082 - mycroft.client.speech.listener:create_wakeup_recognizer:235 - INF                                                      O - creating stand up word engine
11:38:51.085 - mycroft.client.speech.hotword_factory:create_hotword:139 - INFO -                                                       creating wake up
11:38:53.166 - mycroft.client.speech.mic:listen:490 - DEBUG - Waiting for wake w                                                      ord...
`
changing the wakeword back to 'hey mycroft' on the webinterface makes it work with hey mycroft again.

Will probably try building a stretch image again, by following the wiki. Would appreciate a permanent fix, or an option to use snowboy or another working custom wakeword solution,

forslund, 2017-09-29:
Thanks for the update. I would also like a permanent fix :)

Sadly I'm still having trouble recreating the issue on the current dev, I can change to both Hey Pedro and hey Victoria using both web config and home.mycroft.ai

Since it's still not working for you there is some difference that I can't quite put my finger on.

tjoen, 2017-10-04:
So how did you install your raspberry, what image did you use? I would certainly not mind reinstalling, if that would get it to work? I'm wondering how to update with git, since the structure on a pi is a bit different, not using virtualenv for instance....

I also tried installing on raspbian stretch twice, but that seems to break the ssl module in python, and would require building your own python dist. 

I do really appreciate all your effords. Thanks for all your hard work.

forslund, 2017-10-04:
I've been using the Mycroft Mark-1 unit and two desktop installations and not a Picroft which may make a difference but seems unlikely. Going to try that as well.

@Mn0491 did some attempts with upgrading to stretch recently. He hit some snag as well, he's been building a new image that's currently in testing.

tjoen, 2017-10-06:
I managed to install everything on ubuntu mate on a raspberry 3. I cloned the mycroft-core and ran dev_setup.sh. My own compiled mimic was hanging a lot so I removed it and added the dev repo to my sources.list and installed it from there.

Now I can run a git pull, and update the mycroft-core a lot easier.

But if I change the wakework either in the home.mycroft.ai or in the mycroft.conf it still wont pick it up. 
Changin it in the webinterface gives me an interesting warning now though:

`17:03:03.658 - mycroft.client.speech.mic:listen:484 - DEBUG - Waiting for wake word...
17:03:03.683 - mycroft.client.speech.listener:create_wake_word_recognizer:214 - INFO - creating wake word engine
17:03:03.689 - mycroft.client.speech.hotword_factory:create_hotword:134 - INFO - creating hey pedro
17:03:03.697 - mycroft.client.speech.hotword_factory:__init__:53 - WARNING - None module does not match with Hotword class pocketsphinx`

Might help you indebugging this issue?

tjoen, 2017-11-15:
This problem has been fixed since 9.1

closed 2017-11-15 by tjoen