MycroftAI/enclosure-picroft #39: Message Bus stops logging

aussieW, 2017-10-01:
I have noticed on 3 separate picrofts that mycroft-messagebus.log stops capturing messages not long after booting.

Below is the entire mycroft-messagebus.log for a system that has been running for 5 days:

'''2017-09-27 19:09:31,961 - mycroft.configuration - INFO - Loading configuration: /usr/local/lib/python2.7/site-packages/mycroft_core-0.8.22-py2.7.egg/mycroft/configuration/mycroft.conf
2017-09-27 19:09:31,972 - mycroft.configuration - DEBUG - Configuration '/usr/local/lib/python2.7/site-packages/mycroft_core-0.8.22-py2.7.egg/mycroft/configuration/mycroft.conf' loaded
2017-09-27 19:09:31,981 - mycroft.configuration - DEBUG - Configuration '/home/mycroft/.mycroft/mycroft.conf' loaded
2017-09-27 19:09:31,982 - mycroft.configuration - INFO - Loading configuration: mycroft.ai
2017-09-27 19:09:32,006 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
2017-09-27 19:09:33,950 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/device/2679f45e-eb27-45ba-9ade-e1d695cbcfd7/setting HTTP/1.1" 200 2919
2017-09-27 19:09:34,037 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
2017-09-27 19:09:39,265 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/device/2679f45e-eb27-45ba-9ade-e1d695cbcfd7/location HTTP/1.1" 200 936
2017-09-27 19:09:39,356 - mycroft.configuration - INFO - Loading configuration: /etc/mycroft/mycroft.conf
2017-09-27 19:09:39,358 - mycroft.configuration - DEBUG - Configuration '/etc/mycroft/mycroft.conf' loaded
2017-09-27 19:09:39,359 - mycroft.configuration - DEBUG - Configuration '/home/mycroft/.mycroft/mycroft.conf' loaded
2017-09-27 19:09:39,360 - mycroft.configuration - INFO - Loading configuration: /home/mycroft/.mycroft/mycroft.conf
2017-09-27 19:09:39,361 - mycroft.configuration - DEBUG - Configuration '/home/mycroft/.mycroft/mycroft.conf' loaded
2017-09-27 19:09:39,362 - mycroft.configuration - DEBUG - Configuration '/home/mycroft/.mycroft/mycroft.conf' loaded
/var/log/mycroft-messagebus.log (END)```

forslund, 2017-10-07:
Looking at it, the message bus service doesn't have a single LOG statement in itself so it won't ever log much. The log messages are from loading the config at startup and it isn't registered to update configuration when web configuration change.

The process should probably log some more things like if it's auto-restarted, maybe new connections of that's possible with this loose architecture.

What sort of messages would you expect here?

aussieW, 2017-10-07:
Not exactly sure what I should expect there, just figured something was wrong.
I recently installed mycroft on a VM and saw that there are a lots of message-bus messages being logged in mycroft-bus.log so I figured there was something wrong on the picroft version.
Don't want to be missing any helpful info when trying to debug the skills I am developing :).

aussieW, 2017-10-21:
This issue seems to be resolved in the 17.08 release.

closed 2017-10-21 by aussieW