MycroftAI/enclosure-picroft #40: PICroft 0.8.22 version Flawed??

rikMY, 2017-10-03:
I am a newbie here but after installing the image with 0.8.21 and having the auto update bringing it upto 0.8.22 I get strange experiences.  Sometimes the speaker does not work, sometimes the mic does not work, (Jabra 410). Any mycroft.conf changes at any of the 3 conf locations, or via Web always breaks the system completely.
After two weeks of playing around with new images and with the PA and ALSA setting thinking I had a Audio issue, i got nowhere.
Since two days I installed the 'unstable' 0.8.18 image and blocked all updates, It's working perfect, never an issue, even when edit the hard coded mycroft.conf file in the core folder.

Can someone confirm my worries, because i'm quickly losing interest in Mycroft as a concept..

thanks
RM

forslund, 2017-10-03:
Hi @rikMY,

I hadn't heard that things were this bad in 0.8.22 on picroft. There are however a couple of bugs in there. For example, there is one change that will cause Mycroft voice service to crash if using a custom wake word (i.e. not "Hey Mycroft").

Other configuration changes should not cause any major issues. I'd be interested to hear what parameter you're changing so we can figure out other issues.

There's a new picroft image in the pipe with the latest updates and packages that should be an improvement over the existing image.

rikMY, 2017-10-03:
Hi forslund.... the biggest pain was the random audio issue. Either it worked or the mic was dead or the speaker was dead, but even when the speaker was dead it still worked if you used aplay from bash..... 

regarding parameter changes it was just the     "wake_word" and  "phonemes".

anyway as I said I’m so happy with 0.8.18 that I will stay a while and play around with this version before attempting to upgrade again :).

thanks !!

forslund, 2017-10-03:
`wake_word` and `phonemes` are the exact settings that causes issues in 0.8.22, sorry about that.

Stay tuned for the new image, and have fun with 0.8.18 in the meantime :)

rikMY, 2017-10-03:
now I can back to my original issues... trying to figure out if I can make some 'breakout' 2-way conversation by using the 'expect_response=True' :)

Although reading online, I think the 'def converse' method would be better but this is not available yet, is this correct??

forslund, 2017-10-03:
You will probably use `expect_response=True` anyway.

Converse is one of two methods for intelligent conversation.

Converse: Basically overrides intent parsing, allows you to control _everything_. The drawback is that you need to reimplement the logic for parsing sentences.

Conversational Context: Uses the intent parser's context to allow intents to be triggered (or not) depending on what intent was triggered last.

forslund, 2017-10-03:
I think neither of above are available in 0.8.18 :(

rikMY, 2017-10-03:
LOL :)... I was playing with this script on 0.8.21.....then the device auto updated to 0.8.22 and all hell broke loose, so I could not test anymore :D... 

... but on the other side two weeks ago I had never heard of python (or Json),   I just got the urge one day to play around, and after I leant the hard way that Python is sensitive to spacing :), I thought I would try to create a skill... it’s been a uphill struggle ever since.... :)
 
   def handle_next(self, message):
        self.kodi_mode = True
        self.speak_dialog("Kodi.Start", expect_response=True)

def converse(self, utterances, lang="en-us"):
        if self.kodi_mode:
                intent, skill_id = self.parser.determine_intent(utterances[0])
                if skill_id == "favorities":
                        payload = {"jsonrpc": "2.0", "method": "GUI.ActivateWindow", "params": { "window": "favourites"},"id":"1" }
                        response = requests.post(kodi_json_rpc_url, data=json.dumps(payload))
                        self.speak("favorities selected.")
                elif skill_id == "menu":
                        payload = {"jsonrpc": "2.0", "method": "Input.Home"}
                        response = requests.post(kodi_json_rpc_url, data=json.dumps(payload), headers=headers)
                        self.speak("menu selected.")

forslund, 2017-10-04:
Two weeks? then you're making good progress I'd say!

rikMY, 2017-10-06:
Just an update
I let the 0.8,18d  image upgrade to 0.8.22 and I experienced the same issues again with the mic and speaker. So what I did was extract the Mycroft core package from the '64bit' 0.8.21 deb file and played them over the 8.22 files. This solved the sound issues completely,  plus I can still use my own conf file and my wake word to triggers the system. 
This is the good news, only problem is now that the skills don't response to the input, but this is for me to sort out,  :)

penrods, 2017-12-06:
Since you seem to be moving on, I'm going to close this one.

FYI, I'd recommend moving on to the newest 0.9 image (https://mycroft.ai/to/picroft-image).

closed 2017-12-06 by penrods