MycroftAI/enclosure-picroft #41: autorun tries to launch services on every terminal

jherico, 2017-11-28:
The autorun script seems to assume that if you're not connecting over SSH it should launch the mycroft services.  But if you're working locally, it's not uncommon for an experienced *nix developer to rely on multiple TTYs in the absence of a windowing system (ctrl-alt-1, ctrl-alt-2, etc).  

The autorun script should detect if the services have already been launched or detect that we're not running on the default TTY and not attempt to launch them again.

penrods, 2017-11-28:
Are you certain it is relaunching?  The script explicitly checks for SSH logins here
https://github.com/MycroftAI/enclosure-picroft/blob/master/home/pi/auto_run.sh#L21

It only runs the last portion of the autorun (line 164 onward) if not running on the main console.

jherico, 2017-11-28:
I'm not talking about SSH logins.  If you connect the Pi to a monitor and keyboard and boot it up it will be on TTY 1 and will login and run the script.  If you then hit `ctrl-alt-2` it will switch to TTY 2 and give you a login prompt.  Logging in as `pi` will then re-run the script including the content in the if statement, because TTY 2 isn't an SSH login, it's just another terminal on the same system.

penrods, 2017-11-28:
Ah, that isn't a use case the script expected.  Any suggestions to add to the script to catch that?

jherico, 2017-11-28:
you could look at the output of `/usr/bin/tty`  and skip the bulk of the code if it's not `/dev/tty1`

rgreenosborne, 2017-12-11:
Just ran into this problem troubleshooting microphones.  Opened a second local terminal to watch the logs.

Zephael86, 2017-12-20:
I can confirm.
Same use case as @jherico

penrods, 2018-03-15:
Fixed by #44

closed 2018-03-15 by penrods