MycroftAI/enclosure-picroft #43: Reboot problem

Chenel000, 2017-12-20:
Hi, i have installed Picroft v0.8b Rapbian JessieLite on my RPi3, starts correctly and i can pair and register the unit correctly. The problem appears when i reboot the system, the system corrupts and start but with a lot of problems.

Is it necessary to shutdown or stop Mycroft the system in any specific way, i shutdown using "sudo shutdown now".

I have read that in order to recognize the new installed skills is necessary to restart, how can i restart M ycroft without rebooting the system?

MatthewScholefield, 2018-05-15:
Sorry for getting to this so late. If you still are having this problem, would you be able to post the errors you are seeing after a reboot? A standard `sudo reboot` or `sudo shutdown now` should work fine.

penrods, 2018-09-07:
Since there isn't further response, I'm assuming this is no longer an issue.

As a note, SD corruption on a Raspberry Pi is often due to insufficient power.  Beware of the yellow lightening bolts, use at least a 2.4 or 3A power supply!

closed 2018-09-07 by penrods