MycroftAI/enclosure-picroft #54: Error while processing skills

SergioML9, 2018-02-26:
Hi,

I'm running picroft on a Raspberry Pi 3. I don't have any microphone yet, so I'm testing it using the command "say_to_mycroft". The problem is that when I trigger my custom skill, after saying the response that I want, Mycroft says "An error occurred while processing a request in SKILL_NAME skill". I have been reading the logs and I can see this message (which appears also for the pre-installed skills):

`22:01:16.934 - mycroft.skills.main:handle_converse_request:510 - ERROR - Error in converse method for skill 132214865
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/mycroft_core-0.9.17-py2.7.egg/mycroft/skills/main.py", line 504, in handle_converse_request
    result = instance.converse(utterances, lang)
AttributeError: 'NoneType' object has no attribute 'converse'`

Thanks in advance !

nevercast, 2018-02-26:
Looks like an issue in the core. Do you mind posting the issue to https://github.com/MycroftAI/mycroft-core ?

Error line here https://github.com/MycroftAI/mycroft-core/blob/dev/mycroft/skills/main.py#L504
Instance comes from here https://github.com/MycroftAI/mycroft-core/blob/dev/mycroft/skills/main.py#L497
Which comes from the skill.

You'll probably have to show source for your skill, or at minimum, a cut down version so that the issue can be reproduced.

Cheers

SergioML9, 2018-02-26:
Sorry, I had a problem while installing my skill and I had different installations of it. Everything has solved after rebooting. Thanks very much for your early response.

Cheers

closed 2018-02-26 by SergioML9