MycroftAI/enclosure-picroft #58: Wrong msm script

MatthewScholefield, 2018-05-18:
Running an updated version of the latest picroft 3-14 image, `msm` points to `/home/pi/bin/msm` rather than `/usr/bin/msm`. This can be manually fixed with `rm /home/pi/bin/msm`.

jaller94, 2018-07-21:
What is the difference of `/home/pi/bin/msm` and `/usr/bin/msm`?

`/home/pi/bin/msm` provides a nice help but I have no idea what `/usr/bin/msm` does.

```
pi@picroft:~ $ msm search twitter
Possible match: twitter-skill
pi@picroft:~ $ /usr/bin/msm search twitter
mycroft-timer
```

forslund, 2018-07-21:
Hi @jaller94

There are a bunch of differences. The old msm in /home/pi/bin/msm is using the old repo with skills that aren't compatible with the 18.02 release of mycroft. It will also not install the requirements correctly for mycroft to use them.

The new msm in `/usr/bin/msm` uses the correct repo/branch for 18.02, it also locks the skills to the version they were submitted to the mycroft-skills repo and it handles the requirements correctly.

penrods, 2018-09-07:
Fixed by #69

closed 2018-09-07 by penrods