MycroftAI/enclosure-picroft #59: Future Proof Mycroft Updates and Installs by Preallocating 8GB in the Image

skeledrew, 2018-05-19:
This pertains to installing Mycroft via image on a memory card based system. As per [convo with @KathyReid](https://chat.mycroft.ai/community/pl/5hwnmckuapbdmyd3gen6da153o) I observed that after writing the 20180301_build18 image to a 16GB mSD, only 3.9GB is allocated to the main partition (this can be viewed with any partition manager). This is problematic because it is expected that Mycroft will eventually use ~8GB, and eventually a user may encounter difficult to debug issues when attempting to update, installing a skill, etc and there's no more space on the main partition, esp if they may know that they have at least a 8GB card installed. 

This potential issue can be rectified in 2 ways. The first is for users to manually resize the partition themselves, but this may take a bit of know-how (it was pretty easy to do in KDE Partition Manager) and the user would have to be aware that this is the cause. The second method, which I propose, is to preallocate the 8GB within the image itself before distributing. As it now stands, the space used on the partition is ~3.04GB of that 3.9GB allocated. I think it's just a matter of installing Mycroft to a 8GB card, ensuring that the main partition fills it, and `dd`ing that partition - or the entire card - to create the image.

As a bonus, if it doesn't already, one of the things the support skill could collect is disk usage information, or maybe Mycroft could check if it has enough space to upgrade or install X...

penrods, 2018-06-05:
I thought the image automatically expanded on first boot.  I believe I used PiShrink, which is supposed to do this.  Can you verify this?

jaller94, 2018-07-21:
Yes, the main partition expands on the first boot.

closed 2018-09-07 by penrods