MycroftAI/enclosure-picroft #60: “Locale not supported” error when starting mycroft-cli-client

hummelo, 2018-06-15:
Using Raspberry Pi 3 B
using the latest Image and installed with etcher
The issue I am facing is exactly as discussed here:
https://community.mycroft.ai/t/picroft-install-locale-not-supported-error-when-starting-mycroft-cli-client/3242
-------------------------------

> used the test_microphone command to check that my mic is working properly and I got a nice and clean playback.

> Since the voice recognition it was still not working, I started mycroft_cli_client manually after SSHing my raspberry and I got the following error:

> pi@picroft:~ $ mycroft-cli-client
15:11:27.216 - root - INFO - Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
15:11:27.353 - root - INFO - Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
Locale not supported, please try starting the command and setting LANG="en_US.UTF-8"

>         Example: LANG="en_US.UTF-8" ./start-mycroft.sh cli
> I checked and my locales were all en_GB.UTF-8 so I used raspbi-config to generate and set up the system locales to en_US.UTF-8.

forslund, 2018-06-15:
Did that make it work?

```
15:11:27.216 - root - INFO - Generating grammar tables from /usr/lib/python2.7/lib2to3/Grammar.txt
15:11:27.353 - root - INFO - Generating grammar tables from /usr/lib/python2.7/lib2to3/PatternGrammar.txt
```
This looks like an older version running and not the latest version...

hummelo, 2018-06-15:
I will check later on, the image I used is raspbian_Picroft_2018-03-14.zip

forslund, 2018-06-15:
Hmm, that is the one I've used for release tests, it should upgrade to the latest version after wifi is setup...

hummelo, 2018-06-27:
the upgrade works, and picroft is operating ok, but if I connect via ssh to the pi I am still getting the error:

`>22:30:50-d036795@WDFM33987900A:~  $  (minikube) ssh pi@192.168.1.69
pi@192.168.1.69's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Jun 27 20:31:01 2018
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named mycroft.version

***********************************************************************
** Picroft enclosure platform version: 2018-03-14
**                       
***********************************************************************
This image is designed to make getting started with Mycroft easy.  It
is pre-configured for a Raspberry Pi that has a speaker or headphones
plugged in to the Pi's headphone jack, and a USB microphone.
***********************************************************************
Remote session

***********************************************************************
In a few moments you will see the contents of the speech log.  Hit
Ctrl+C to stop showing the log and return to the command line.  Mycroft
will continue running in the background for voice interaction.

Additional commands you can use from the command line:
    mycroft-cli-client - command line client, useful for debugging
    msm                - Mycroft Skills Manager, install new Skills
    say_to_mycroft     - one-shot commands from the command line
    speak              - say something to the user
    test_microphone    - record and playback to test your microphone
***********************************************************************

20:38:50.788 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /opt/venvs/mycroft-core/lib/python3.4/site-packages/mycroft/configuration/mycroft.conf loaded
20:38:50.955 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /opt/venvs/mycroft-core/lib/python3.4/site-packages/mycroft/configuration/mycroft.conf loaded
20:38:51.126 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /etc/mycroft/mycroft.conf loaded
20:38:51.294 - mycroft.configuration.config:load_local:129 - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' not found
20:38:51.312 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
20:38:51.865 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/device/ HTTP/1.1" 301 106
20:38:51.876 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTP connection (1): api.mycroft.ai
20:38:52.123 - requests.packages.urllib3.connectionpool - DEBUG - http://api.mycroft.ai:80 "GET /v1/device HTTP/1.1" 302 0
20:38:52.129 - requests.packages.urllib3.connectionpool - DEBUG - Resetting dropped connection: api.mycroft.ai
20:38:52.684 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/device HTTP/1.1" 401 38
20:38:52.704 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
20:38:53.257 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/auth/token HTTP/1.1" 401 38
20:38:53.436 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /var/tmp/mycroft_web_cache.json loaded
20:38:53.606 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /etc/mycroft/mycroft.conf loaded
20:38:53.770 - mycroft.configuration.config:load_local:129 - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' not found
Locale not supported, please try starting the command and setting LANG="en_US.UTF-8"

	Example: LANG="en_US.UTF-8" ./start-mycroft.sh cli

22:30:50-d036795@WDFM33987900A:~  $  (minikube) ssh pi@192.168.1.69
pi@192.168.1.69's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Jun 27 20:31:01 2018
Traceback (most recent call last):
  File "<string>", line 1, in <module>
[... 22 lines cut ...]
    test_microphone    - record and playback to test your microphone
***********************************************************************

20:38:50.788 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /opt/venvs/mycroft-core/lib/python3.4/site-packages/mycroft/configuration/mycroft.conf loaded
20:38:50.955 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /opt/venvs/mycroft-core/lib/python3.4/site-packages/mycroft/configuration/mycroft.conf loaded
20:38:51.126 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /etc/mycroft/mycroft.conf loaded
20:38:51.294 - mycroft.configuration.config:load_local:129 - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' not found
20:38:51.312 - requests.packages.urllib3.connectionpool - DEBUG - Starting new HTTPS connection (1): api.mycroft.ai
20:38:51.865 - requests.packages.urllib3.connectionpool - DEBUG - https://api.mycroft.ai:443 "GET /v1/device/ HTTP/1.1" 301 106
[... cut ...]
20:38:53.436 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /var/tmp/mycroft_web_cache.json loaded
20:38:53.606 - mycroft.configuration.config:load_local:124 - DEBUG - Configuration /etc/mycroft/mycroft.conf loaded
20:38:53.770 - mycroft.configuration.config:load_local:129 - DEBUG - Configuration '/home/pi/.mycroft/mycroft.conf' not found
Locale not supported, please try starting the command and setting LANG="en_US.UTF-8"

	Example: LANG="en_US.UTF-8" ./start-mycroft.sh cli

`

>

Schoonology, 2018-06-29:
I'm running into this error this morning with the latest build. The detail that I'll add is that connecting via SSH seems to have _triggered_ the error, but connecting locally from then on seems to have the same `Locale not supported` problem.

forslund, 2018-06-29:
Some (all) ssh clients seem to send locale when connecting to a host. It might be this that triggers the issue. There is a workaround for that detailed here: https://stackoverflow.com/questions/29609371/how-do-not-pass-locale-through-ssh

can you run `locale`

and paste the output here?

The pi has the GB locale installed by default so it might work to launch the cli setting the LANG to en_GB.UTF-8:

`LANG=en_GB.UTF-8 mycroft-cli-client`

Schoonology, 2018-06-29:
I added an `export` to the `pi` user's `.bashrc` file to set the `LANG` environment variable for all login shells, so `locale` displays the following:

```
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE=en_US.UTF-8
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

Schoonology, 2018-06-29:
Running neither of these works, either (checked over SSH):

```
pi@picroft:~ $ LANG="en_GB.UTF-8" mycroft-cli-client
pi@picroft:~ $ LANG="en_US.UTF-8" mycroft-cli-client
```

Schoonology, 2018-06-29:
Thanks for the suggestions, @forslund. To round out the information I might hope to provide, it looks like not accepting `LANG` _does not_ resolve the issue.

I'm happy to try any other suggestions.

forslund, 2018-06-30:
The errors when running locale shows that the en_US locale isn't installed (which is expected)

what's the output of
`LANG="en_GB.UTF-8" locale`

because that's a locale that should be installed.

Schoonology, 2018-07-01:
With a little more futzing (and plenty of device restarts), it looks like the fix you linked earlier _does_ resolve the issue. In `/etc/ssh/sshd_config`, the following gets commented out:

```
# Allow client to pass locale environment variables
# AcceptEnv LANG LC_*
```

Going forward, should the Picroft enclosure provide this configuration when it's set up? Otherwise, should the error message and suggestion be updated to use `en_GB`, instead, or to find an installed locale?

penrods, 2019-07-22:
I don't believe this is an issue anymore.  Let me know if you still see this.

closed 2019-07-22 by penrods

kallisti5, 2020-05-02:
Still an issue on latest master installed to Raspian on a Raspberry Pi 4.
```
$ mycroft-cli-client 
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/mycroft-core/mycroft/client/text/__main__.py", line 21, in <module>
    from .text_client import (
  File "/home/pi/mycroft-core/mycroft/client/text/text_client.py", line 39, in <module>
    locale.setlocale(locale.LC_ALL, "")  # Set LC_ALL to user default
  File "/usr/lib/python3.7/locale.py", line 604, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting

$ locale
.
.
LC_ALL=
.
```

This fixes it:
```
LC_ALL=C mycroft-cli-client
```