MycroftAI/enclosure-picroft #61: configure wifi doesn't work with Mycroft core >= 18.2.6

clebio, 2018-06-15:
I've been trying to make a new image of picroft that works with the Google AIY voicekit, but am having issues with the wifi setup.

On a Raspberry Pi 3 B, with 
* mycroft-core 18.2.8
* mycroft-picroft 18.2.8
* mycroft-wifi-setup 0.2.5

`source configure_wifi.sh` fails, saying 
`ImportError: No module named mycroft.messagebus.client.ws`

This seems clearly due to the fact that the shell script is using python2:

```
ls -l `which python`
/usr/bin/python -> python2.7
```
https://github.com/MycroftAI/enclosure-picroft/blob/master/home/pi/configure_wifi.sh#L2

Except `messagebus_emit.py` is not compatible with Python3:
```
python3 messagebus_emit.py
...
SyntaxError: Missing parentheses in call to 'print'
```
https://github.com/MycroftAI/enclosure-picroft/blob/master/home/pi/messagebus_emit.py#L12

Presumably the error started with mycroft-core v18.2.6, 
> The big move to python 3 ...
https://github.com/MycroftAI/mycroft-core/releases/tag/release%2Fv18.2.6

Do you have a feature branch in the works for this resolution? Any workarounds or mitigations you'd recommend for now?

forslund, 2018-06-17:
Thanks for flagging this, we still need to update the Picroft scripts.

I'm pulling this from my head but you should be able to do replace that script with

```
#! /opt/venvs/mycroft-core/bin/python

from mycroft.messagebus.send import send

send('mycroft.wifi.start')

print("=====================================================================")
print("Wifi setup has begun.  Use your phone, tablet or laptop to connect to")
print("the network 'MYCROFT' using the password '12345678'.  Once connected")
print("browse to 'https://start.mycroft.ai', then follow the prompts to")
print("complete the setup.")
print("=====================================================================")
```

clebio, 2018-06-17:
Thanks for this. I replaced the line in `/home/pi/auto_run.sh` and tested on a new image that I built. Seems like I ran into an (unrelated?) issue with `precise-stream` getting hung, though a reboot resolved that.

closed 2018-06-17 by clebio