MycroftAI/enclosure-picroft #63: auto_run.sh attempts to check mycroft version from ouside the virtual environment

ScottJWalter, 2018-07-09:
## Stack:
* Raspberry Pi 3
* 8GB MicroSD card
* Picroft (core 18.2.10)

## Steps to Replicate
1.  Install and Mycroft on the RPi 3
2.  Configure SSH access to the RPi 3 (eg. via raspi-config)
3.  SSH into the RPi 3

## The Problem

When logging into the RPi 3 via SSH, the following header is displayed:

> $ ssh pi@picroft
>
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> Last login: Mon Jul  9 12:19:25 2018
> **Traceback (most recent call last):**
>   **File "<string>", line 1, in <module>**
> **ImportError: No module named mycroft.version**
> 
> ***********************************************************************
> ** Picroft enclosure platform version: 2018-03-14
> **
> ***********************************************************************

Quick inspection of the `auto_run.sh` script points to line 12:

`mycroft_core_ver=$(python -c "import mycroft.version; print 'mycroft-core: '+mycroft.version.CORE_VERSION_STR" | grep "core:")`

Which is being run _outside_ the mycroft virtual environment, so it can't see the mycroft Python packages.

forslund, 2018-07-09:
Thanks for reporting. This should be fixed in PR #62 and will be merged shortly (I hope).

closed 2018-09-07 by penrods

ali97hz, 2018-11-07:
same error