MycroftAI/enclosure-picroft #66: Picroft2:  Add AYI support

penrods, 2018-08-03:
For the new Picroft setup-wizard, we can add support for the original AIY kit using these instructions:

```
Don't know if this is still relevant, but this might be what you were looking for: https://community.mycroft.ai/t/mycroft-with-google-aiy-voice-kit-disk-image/2607/11
and here is the support for the AIY button & LED https://gist.github.com/tjoen/9851436cedb68f6d6a08be14067c85f4 (I think the code is Python 2.7)
```

JarbasAl, 2018-08-27:
i am using the google kit in a git install, this is what i did

# installing aiy

echo "deb https://dl.google.com/aiyprojects/deb stable main" | sudo tee -a /etc/apt/sources.list.d/aiyprojects.list
wget -q -O - https://dl.google.com/linux/linux_signing_key.pub | sudo apt-key add -
sudo apt-get update
sudo apt-get upgrade
sudo reboot

sudo apt-get install aiy-dkms aiy-voicebonnet-soundcard-dkms aiy-voicebonnet-routes
sudo apt-get install aiy-python-wheels
sudo apt-get install leds-ktd202x-dkms

# install GPIO in .venv

source "${VIRTUALENV_ROOT}/bin/activate"

pip install RPi.GPIO

# code

here is my enclosure code for it, needs https://github.com/MycroftAI/mycroft-core/pull/1306 or the most recent https://github.com/MycroftAI/mycroft-core/pull/1760 should also work
```

from mycroft.enclosure import Enclosure
from mycroft.audio import is_speaking
from mycroft.messagebus.message import Message
from mycroft.skills.audioservice import AudioService
from mycroft.enclosure.aiy.voicehat import get_button, get_led

class GoogleAiyVoiceEnclosure(Enclosure):
    def __init__(self):
        Enclosure.__init__(self)
        self.audio = AudioService(self.bus)
        self.button = get_button()
        self.led = get_led()

        self.button.on_press(self.handle_button_press)

    def handle_button_press(self):
        if is_speaking():
            self.bus.emit(Message("mycroft.stop"))
        elif self.audio.is_playing:
            self.audio.stop()
        else:
            self.bus.emit(Message("mycroft.mic.listen"))

    def shutdown(self):
        Enclosure.shutdown(self)
        self.button.on_press(None)

```

penrods, 2018-09-10:
@JarbasAl What version of Raspbian are you using?  Jessie or Stretch?  I'm looking at adding this to the new script and I'm thinking I can probably skip the "apt-get upgrade" on Stretch.

JarbasAl, 2018-09-13:
i used stretch

one detail, i included the python package in the source code (so i could explore the source in my favorite IDE) and forgot that install step

https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md#python-library

this step is missing from the above

andlo, 2018-09-15:
I think I did make correct changes to auto_run.sk so i now install google AIY. I also made som aditional changes so it worked :) both sound and mic - and soundlevel as well.
I made a PR, but as it is my first one i am not sure i did it right ;I

by the way the trick to get AIY installed were 
`sudo mkdir /usr/lib/systemd/system `
before installing aiy-io-mcu-firmware.

To get sound and mic working i aded a asound.conf and made changes to /etc/mycroft/mycroft.conf

All of it i added to the auto_run.sh

andlo, 2018-09-17:
@JarbasAl 
> here is my enclosure code for it, needs MycroftAI/mycroft-core#1306 or the most recent MycroftAI/mycroft-core#1760 should also work

How do one use the enclosure code ? where to put it and use it ?

penrods, 2018-09-18:
After the base image is working, I would like to build a generic Pi GPIO skill that implements button support.  I can easily see using the requirements.txt to install RPi.GPIO and allow support of one or more buttons on various GPIOs -- I'm seeing a web UI that allows the user to pick whether the button acts like the Mark 1, act as Volume Up, act as Volume Down, act as Mute, etc.

But I want this to be done as a Skill that is tagged to support Picroft and/or Mark 1, not as part of the Picroft image itself.  Users will pull that down from the Marketplace and we might go ahead a make it a default Picroft skill.

penrods, 2019-01-29:
I believe #72 and #85 takes care of the original reason for this issue -- AIY mic support.  I'm going to go ahead and close.  The button support can be dealt with in a Skill.

closed 2019-01-29 by penrods