MycroftAI/enclosure-picroft #73: Picroft2: pulseaudio or not...

andlo, 2018-09-15:
I have installed new image, and got google AIY working.

But if installing pulseaudio mycroft cant set volume. If not having installed pulseaudio he can set volume.

Cant figure out why and how to fix.

I think I need pulse audio to get mix on google AIY working whith mycroft. It works whitout pulseaudio outside mycroft - and whith mycroft-mic-test but non when mycrofr is running (gues the voice part dosnt really work vhitout pulse)

andlo, 2018-09-16:
The mic does show there also if not using pulse - it starts by changing level and stops. 

Where can I check default play_wav ? 

I did change in mycroft.conf line for aplay and mpg123 to get YIY voicehat working, but that is regarding sound. The problem is mic is hanging when not using pulse.

forslund, 2018-09-16:
If it's changed from the default there should be an entry in the `/etc/mycroft/mycroft.conf` file. or `~/.mycroft/mycroft.conf`

andlo, 2018-09-16:
OK - so picroft is not supposed to use pulseaudio? I just cant get him to function whitout :( problems whit mic is hanging. 
Testing mic whith arecord and aplay is working altså whitout pulse.

andlo, 2018-09-17:
Yes the only issue using pulseaudio is volume skill dosnt function - or it does, and it changes volume which can be seen by alsamixer, but sound output is not afected.
read somewhere that there is a issue (or were in earlier dayes) whit volumelevel and raspbian when using pulseaudio :( 

Strange thing is mic and volume did work fine on picroft v.1 image (the one based on jessie) 

The new picroft image dosnt have pulseaudio installed by default - I dont know if the old one did.
Maybe pulseaudio should be in the new image by default  ?

penrods, 2019-01-29:
#85 deals with this, closing

closed 2019-01-29 by penrods