MycroftAI/enclosure-picroft #82: Creating a raspbian full build for development purposes

congruentsquare, 2018-11-02:
I am running into a bunch of issues with the installation script due to different configs(or something else). Many people need a gui for development purposes and was wondering if there could be a full raspbian(non-lite) image release with picroft? It would greatly simplify mycroft development on the pi.

andlo, 2018-11-29:
I am nearly finish on a skill that do install a compleate IDE with all the stuff needed to make skills etc. for mycroft. The IDE is installed localy on the device, and can be accesed throu a webbrowser. 

https://github.com/andlo/theia-ide-skill

closed 2019-07-12 by penrods

jshep321, 2019-11-23:
Hi @penrods ,
Just a suggestion, but I think this (full desktop + picroft) should be a target for the buster image given the horsepower on the pi4.  

Personally, I'm envisioning a custom skill that shows photos/artwork  (chromium browser in kiosk mode) when idle, and mopidy/spotify when active.  

Thoughts?