MycroftAI/enclosure-picroft #89: Unable to install Mycroft over RPI3+ AIY kit

nciervo, 2019-01-03:
I'm trying to get Mycroft to work over the Magpi AIY kit (v1). 

I tried using the the 09-12 stable image and Andlo's [wget](https://community.mycroft.ai/t/picroft-2018-09-12-and-the-google-aiy-voice-hat/4725/15)s. Unfortunately, after the "4) for Google Voice hat", the installation stalls on that same exact step the next go around.

I reloaded the fresh image, tried again without rebooting, but got the [protobuf](https://community.mycroft.ai/t/setting-up-aiy-python-wheels-protobuf-not-supported-on-armv6-1/5130) error. I tried using the install_AIY.sh script Andlo prepared (with AIY-python-wheels) commented out, but my system seemed to still want to try and install them (even after removing and reloading the apt repository list multiple times).

I also tried using the latest AIY image and installing Mycroft over top of that image as described in [these instructions](https://github.com/MycroftAI/enclosure-picroft/blob/stretch/image_recipe.md), but upon reboot, it went straight into the OS GUI and MyCroft didn't start on it's own.

Thoughts? Ideas?

# How to submit an Issue to a Mycroft repository

When submitting an Issue to a Mycroft repository, please follow these guidelines to help us help you. 

## Be clear about the software, hardware and version you are running

For example: 

* I'm running a Mark 1
* With version 0.9.10 of the Mycroft software
* With the standard Wake Word

## Try to provide steps that we can use to replicate the Issue

For example: 

1. Burn the 0.9.10 image to Micro SD card using Etcher
2. Seat the Micro SD card in the RPi 3
3. Boot Picroft
4. Wait 3 minutes
5. The red light will come on indicating that the RPi 3 is overheating
6. Running `htop` via the command line indicates a number of Zombie'd processes

## Be as specific as possible about the expected condition, and the deviation from expected condition. 

This is called _object-deviation format_. Specify the object, then the deviation of the object from an expected condition. 

Example 1: 

* When I say "Hey Mycroft, set your eyes to cadet blue", the eyes turn purple instead of blue. 

Example 2: 

* When I say "Hey Mycroft, what time is it in Paris", the time spoken is out by one hour - it's not observing daylight savings time. 

Example 3: 

* When I run `msm default` on my Mark 1, I receive lots of Git 'locked file' errors on the command line. 

## Provide log files or other output to help us see the error

We will normally require log files or other troubleshooting information to assist you with your Issue. 

This [documentation](https://mycroft.ai/documentation/troubleshooting/) explains how to find log files. 

As of version 0.9.10, the [Support Skill](https://github.com/MycroftAI/skill-support) also helps to automate gathering support information. 

Simply say: 

* "Create a support ticket" _or_
* "You're not working!" _or_
* "Send me debug info"

and the Skill will put together a support package which you can email to us. 

## Upload any files to the Issue that will be useful in helping us to investigate

Please ensure you upload any relevant files - such as screenshots - which will aid us investigating.

krisgesling, 2019-02-15:
Hi @nciervo,

Apologies that I didn't see your issue until now. Are you still having the same trouble?

nciervo, 2019-03-24:
Spent most of the morning trying to get it working with the Picroft Stable 2018-09-12 image and [Andlo](https://community.mycroft.ai/t/picroft-2018-09-12-and-the-google-aiy-voice-hat/4725/15)'s updated auto_run.sh and AIY-asound.conf.
When I first let the auto_run go, I selected option 4 for the AIY VoiceHat installation. When prompted by the installer to restart, I did.
The next time around, I selected 4, again and it froze. After 15 minutes I hit ctrl+c, went into the mycroft-core folder and tried starting Mycroft manually. It told me to run the dev-setup.sh again. I tried that and I got the protobuf error. I tried following the directions [here](https://community.mycroft.ai/t/setting-up-aiy-python-wheels-protobuf-not-supported-on-armv6-1/5130/7) to fix it, but couldn't get the protobuf error to go away.
Finally I ran "./dev_setup.sh -n" telling it to skip errors instead of quitting. This allowed me to start MyCroft, but it couldn't hear me and I couldn't hear it.

I will try installing MyCroft from the latest[ AIYProjects image](https://github.com/google/aiyprojects-raspbian/releases) from 2018-11-16 the day that I have free.

andlo, 2019-03-24:
I am looking for a spare sd card to use, and will then flash a new picroft with AIY to se how it goes. Last time (now some time ago) i worked, but something could be broken.

andlo, 2019-03-26:
Just reflashed and dosnt have have any problems....What I did was:

Downoaded the stable image
Flashed and booted up
ssh pi@picroft
Ctrl+C to enter the shell
run sh ./update.sh to update picroft (the autorun.sh)
exit the ssh
ssh pi@picroft
followed the guide and choose 4 for Google AIY
/lots of updating and downloads)
R for reboot
ssh pi@picroft
follow the guide choose 4 for Google AIY
paired when asked
hey mycroft what version are you running to update to latest mycroft-core
hey mycroft install google aiy voice kit skill
hey mycroft install auto volume skill

But I doo have other issue where mycroft isnt speaking last part of long sentences, but tha is a hole other problem.....

penrods, 2019-03-26:
> But I doo have other issue where mycroft isnt speaking last part of long sentences, but tha is a hole other problem.....

I suspect this is something that @Ruthvicp identified and which should be fixed in the next release.

Ruthvicp, 2019-03-27:
Mimic2 long sentences are handled in [this](https://github.com/MycroftAI/mycroft-core/pull/2061) PR.  @andlo Please try with the latest dev branch & let us know if you find any error

andlo, 2019-03-27:
Yep I know. I were concentrating on tis issye where @nciervo have problems installing picroft and getting the AIY installed. That were why I did a reflash to see how things were dooing and it seems to work for me.
Well - my mi is just a pi3 and not a pi3+

nciervo, 2019-03-31:
Following @andlo 's directions, I was about to get through the installation process, Mycroft can hear me, but I cannot hear it.
After the second time of choosing "4 for Google" I was able to hear the microphone test. However, as Mycroft started up there were a few lines of:
     "amixer unable to find simple control 'pcm' 0"
     "Unable to find the given element from control default"

A few differences from what @andlo 's - as he said I am using RPi3+. Also, instead of SSH, I connected directly to the device. Whenever his instructions recommended disconnected and reconnecting, I did "sudo shutdown -r now" -> I doubt this was a different maker. 
Instead of using ssh, I was connected directly to the device

nciervo, 2019-04-16:
I gave it another go this morning and it seems to be working. When I ask it
its IP address it says "192.1" and stops - which seems to be related to the
other issue mentioned in this thread.
But the main issue is solved. I must have been missing a step somewhere. I
apologize for wasting everyone's time.

Fonsos, 2019-04-26:
> Just reflashed and dosnt have have any problems....What I did was:
>
> Downoaded the stable image
> Flashed and booted up
> ssh pi@picroft
> Ctrl+C to enter the shell
> run sh ./update.sh to update picroft [...]

When doing the sh ./update.sh I got this:
`pi@picroft:~ $ sh ./update.sh
Would you like to install Picroft on this machine?
Choice [Y/N]: ./update.sh: 31: read: Illegal option -N
Aborting install.`

penrods, 2019-07-12:
I just tried the new unstable image (https://mycroft.ai/to/picroft-unstable) on my AIY v1.  It seems to work, but I would appreciate another giving it a shot.

tuck182, 2019-07-19:
I just did a first-time install on a new AIY platform yesterday using the new unstable image, and ran into a few issues:

1. Got stuck in a reboot loop with the AIY hardware selection because of https://github.com/MycroftAI/enclosure-picroft/issues/109 (https://github.com/MycroftAI/enclosure-picroft/pull/110 would resolve this).

1. The error `E: Unable to locate package aiy-voicebonnet-routes`
   from https://github.com/MycroftAI/enclosure-picroft/blob/stretch/home/pi/auto_run.sh#L314

1. Compile failure from installing `aiy-dkms` (see [make-aiy-dkms.log](https://github.com/MycroftAI/enclosure-picroft/files/3412188/make-aiy-dkms.log))

1. Compile failure from installing `aiy-voicebonnet-soundcard-dkms` (see 
[make-aiy-voicebonnet-soundcard-dkms.log](https://github.com/MycroftAI/enclosure-picroft/files/3412193/make-aiy-voicebonnet-soundcard-dkms.log))

penrods, 2019-07-22:
I addressed the reboot boneheadness.  I was able to successfully get it running on an AIY using this image, although I'll admit that I walked away during the install process so there might be some warnings I missed.

For now I'm pushing this out as both stable and unstable (since the old stable was nearly a year old).  If this needs modifications we can iterate from here, but I also want to attempt to move to Raspbian Buster very soon.

closed 2019-07-22 by penrods