MycroftAI/enclosure-picroft #98: auto_run.sh from stretch branch does not check out any valid branch

fractal13, 2019-05-11:
I'm setting up picroft using the raspbian-stretch image linked from the picroft page. Notably, the image link is https://mycroft.ai/to/picroft-image.

When auto_run.sh is asked to help configure the system, the two branches you can choose from are masterll and dev.  Is "masterll" used because the stretch branch isn't finished yet?  The other option is "dev", which also does not exist.  So, no matter what a user selects, they will receive an error message from the git checkout command.  Is this be design, or can we do something to make the user experience better?

krisgesling, 2019-05-15:
Hi fractal, great spotting. Not sure how that one snuck in there, just a typo I think. 

Did you want to make the change and issue a pull request to git it fixed up? We can make the change if you don't care, this just recognises your work in finding the problem and contributing it back. 

If you did want to make the change, we would need you to sign our [Contributor License Agreement](https://mycroft.ai/cla). This is primarily to make it crystal clear whether you are contributing as an individual or through employment with some organisation, and that you retain ownership of anything you contribute but provide Mycroft the rights to use it.

fractal13, 2019-05-15:
Submitted pull request just now.  Thanks for your efforts.

krisgesling, 2019-05-16:
Great, will wait for Steve to verify receipt of the CLA, but it all looks good :)

penrods, 2019-05-16:
I did accept this since the Picroft repo doesn't technically require a CLA.  However I did not receive a CLA request if you did submit one.   Let me know if you tried to use https://mycroft.ai/cla -- we've heard a few reports of people requesting the CLA to sign but it didn't go through for some reason.

closed 2019-05-16 by penrods

fractal13, 2019-05-17:
I resubmitted a CLA request.  This time with a different email address. Hope it works.  Thanks for your work on this project.

krisgesling, 2019-05-18:
Hey @fractal13, I want to try and track down what's going on with our CLA form so we don't miss them in the future. Would you be able to email me with your two email addresses and what browser + OS you use?
kris . gesling at mycroft . ai

krisgesling, 2019-05-18:
Hey I've just made some configuration changes, can you try the [CLA form](https://mycroft.ai/cla) one more time?

penrods, 2019-05-20:
Looks like it came through finally!

Steve Penrod
Mycroft.AI <https://mycroft.ai/> | Chief Technology Officer | 816.674.6583
<816-674-6583>
300 E 39th Street Kansas City, MO 64111