MycroftAI/enclosure-picroft #99: Picroft images are outdated

pbajno, 2019-05-15:
Password for current google drive images is mycroft or raspberry?
Could you update those images? They are almost one year old. That's huge.

krisgesling, 2019-05-16:
Hi pbajno, the default password is mycroft. 

We are aiming to release a new image for 19.02, we just have a few other priorities at the moment, so need to find some time to dedicate to this.

penrods, 2019-07-12:
You can try the new unstable image at: https://mycroft.ai/to/picroft-unstable

Once I'm comfortable that it is up to snuff, this will be promoted to stable.  But I'm pretty sure it is solid right now.

penrods, 2019-07-22:
Both  https://mycroft.ai/to/picroft-unstable and  https://mycroft.ai/to/picroft-image are pointing to the latest and greatest now!  This is a 19.2.13 install, by default, and will auto-upgrade.

closed 2019-07-22 by penrods