MycroftAI/enclosure-picroft #104: ustable build disk image is corrupt

hobson, 2019-07-07:
I'm using BalenaEtcher Version 1.5.51 to write the unstable "raspbian-stretch_Picroft_2018-12-17.img.zip" file to a microsd card.  
The Picroft docs section ["downloading the disk image"](https://mycroft.ai/documentation/picroft/#downloading-the-disk-image) provides the unstable "Picroft disk image" link as https://mycroft.ai/to/picroft-unstable .   
This link redirects to google drive at https://drive.google.com/uc?id=1qGE3dlQTc8WLFTSwVMDjKOuneksc45i-&export=download .  
When BalenaEtcher finishes flashing this image to the SD card it complains that **the image file is corrupt**.  And the image does not run on my raspberry pi.
I had no trouble downloading and flashing the older stable version (raspbian-stretch_Picroft_2018-09-12.zip).   

## Issues/Questions?

1. Is there an md5 signature that can be used to verify correct downloading of disk image zip files?
2. Is there a list of all the disk images you've create? The github Releases don't seem to contain the disk image files for each release.
3. How can I create the latest disk image from source? I need the Google AIY Voice kit support only available in the latest unstable release.

forslund, 2019-07-07:
Hi I transfered this issue to the picroft repository, this is basically the overlay that's used to create the pircrof image I'm not sure if all steps are in here though. I'll ping @penrods who (has led the creation of the picroft image) hopefully he can fill in any missing detail.

Currently there's no md5-sum but I'll see if I can produce one.

forslund, 2019-07-10:
Hi, the md5sum is 7f43ac1a3fc53b4f79d90e0374a9e50a.

@krisgesling reports that he's experienced issues when using the compressed image directly with Balena, but the uncompressed image has worked. So try unzipping the archive and flash the sd-card with that instead.

penrods, 2019-07-11:
Can you try the https://mycroft.ai/to/picroft-unstable image now and see it that works for you?  I updated it last night.

The SHA256 listed for the unstable release here (https://github.com/MycroftAI/enclosure-picroft#picroft---2019-07-10-stretch-lightning-release) is correct.  I was a little premature in changing the name associated with the stable release as it is still the old one from last year, but its SHA is correct.

penrods, 2019-07-20:
Since several have downloaded and used the unstable image, I'm considering this closed.

closed 2019-07-20 by penrods