MycroftAI/enclosure-picroft #109: jq parse error in auto_run.sh during initial setup steps

el-tocino, 2019-07-19:
## Be clear about the software, hardware and version you are running

using latest picroft image on pi3 with aiy hat. When connecting to initial few connects, an error occurs after the aiy setup is run. It appears just before the giant mycroft banner is displayed.  

```
parse error: Expected another key-value pair at line 4, column 1
parse error: Expected another key-value pair at line 4, column 1
parse error: Expected another key-value pair at line 4, column 1
```

The issue seems to be the jq parsing that happens at the start of auto_run.sh.  My .setup_config file had the following:
```
{
  "audio": "google_aiy",
  "setup_stage": "aiy_reboot",
}
```

At least one other person is reporting something similar. 

## Try to provide steps that we can use to replicate the Issue

1. Boot a fresh image of Picroft 
2. choose guided setup, let the updates occur.
3. after that first reboot, reconnect, then choose 4, google aiy hat.  Let that setup complete and the reboot occur. 
4. when connecting again, the above error shows up.  

## Be as specific as possible about the expected condition, and the deviation from expected condition. 

Expected the aiy setup to complete after the reboot.  

## Provide log files or other output to help us see the error

n/a, really.  

## Upload any files to the Issue that will be useful in helping us to investigate

included inline.

forslund, 2019-07-19:
This seems to be a trailing `,` issue on the line

`  "setup_stage": "aiy_reboot",`

penrods, 2019-07-20:
Well, dang.  :-/

Nice catch!

penrods, 2019-07-20:
I'll get a new unstable image built ASAP.

PR #110 (2019-07-20): Build setup_choices using jq
This fixes an error with a trailing ',' and makes the code easier to extend.

Resolves #109
closed 2019-07-20 by penrods