MycroftAI/enclosure-picroft #111: Picroft is not connecting to personal or enterprise WiFi

rajdpandey, 2019-07-22:
Hi ,

I am running a 2018 Picroft stable version and till now it was perfectlt connecting with my personal or Enterprise WiFi but I don't know What happened it stoped.

 When I reboot or shutdown and start it stops at Wifi connection wizard. I have checked the wpa_supplicant file I have there my passwords and ssids so what is the problem?

This error was given when I entered again SSID and password: 

Failed to connect to non-global ctrl_ifname: wlan0 error: no such file or directory

penrods, 2019-07-22:
I'm afraid this doesn't sound quite like a Picroft issue, but rather a wifi configuration or hardware problem.

When Picroft starts up it will attempt to verify a live network connection, running the wifi setup if none is found.  Does your network come up slowly?  E.g. if you Ctrl+C out of the setup and run `ping 1.1.1.1` after a little bit, is the network live?  If so, maybe we need to alter the auto_run (although I think it will auto-exit if the network comes up even after the initial test).

If this isn't your issue, then it sounds like there is something weird in the network configuration at the OS level.  Googling that error gave a few suggestions, such as verifying that you don't have spaces around the SSID and password (https://www.raspberrypi.org/forums/viewtopic.php?t=212134), making sure you have the right country code (as not all countries use the same wifi channels), etc.

rajdpandey, 2019-07-22:
Raspberry Pi's wifi chip is not working that's why I was having that issue.

closed 2019-07-22 by rajdpandey