MycroftAI/enclosure-picroft #113: Unable to re-enter setup-wizard

krisgesling, 2019-07-31:
After completing an initial set up of Picroft, the `mycroft-setup-wizard` command is intended to re-run the setup-wizard. This successfully runs however does not prompt for any changes as the script detects that selections have already been set. Instead the command simply checks for updates and then launches the CLI.

My understanding is that the command should allow a complete re-run of the setup-wizard enabling users to select new input/output devices etc.

I'm assuming a recent update to the setup script is preventing a proper re-run of the wizard.

rajdpandey, 2019-09-24:
so is this issue solved yet or still open?

krisgesling, 2019-09-24:
I haven't had time to look at it, so assuming it's still an issue. Have added a help wanted flag to see if anyone in the broader Community is interested in taking a look.

penrods, 2019-10-20:
I believe this is solved in the Buster image.  I have it wipe out the options file:  https://github.com/MycroftAI/enclosure-picroft/blob/buster/home/pi/auto_run.sh#L726

krisgesling, 2019-10-21:
Yeah, can confirm that, I re-entered setup a few days ago. Closing.

closed 2019-10-21 by krisgesling