MycroftAI/enclosure-picroft #114: Raspberry Pi 4 Support

Teutonic84, 2019-08-06:
Does Picroft support the raspberry pi 4b model? And if not, are there plans in the near future to add support for it? I'm heading to a local shop to pick up a pi today and wanted to figure out which one I should go for to use with picroft.

krisgesling, 2019-08-07:
Hi there, you can install Mycroft using the git clone method, however the Picroft image doesn't yet support RPi 4. Picroft is currently built on Debian Stretch rather than Buster which is required for the new models.

It is certainly an update we will get to however is not an immediate priority and is more involved than just running `dist-upgrade` unfortunately :slightly_frowning_face:

andlo, 2019-08-24:
I just got a Pi4 and are looking at making a Picroft image based on raspbian bustner. I hope bigest issues to face is the time it takes ;)

andlo, 2019-08-25:
I did follow the recipe and made a picroft image build on Bustner. It went straight away only with minor differenceses from the recipe. 
@krisgesling should I fork the repo, add a bustner branch, change the recipe and make a image ? Or is there some other plans gooing on like making picroft more like MarkII ?

forslund, 2019-08-25:
@andlo, I created a buster branch. It would be really cool if you could create a PR for the recipe changes to that branch.

If you create an image we could host it for you if you like.

We will probably do some sort of release of the mark-2-pi platform we've been working on during the summer in some form similar to the picroft but the changes you've made would probably come in handy when making that.

andlo, 2019-08-28:
I will do it as soon as I get the time :)

forslund, 2019-08-28:
Great, many thanks @andlo

penrods, 2019-10-20:
I have created a preliminary Picroft based on Buster and tested it on a Pi 4.  You can give it a shot by downloading https://mycroft.ai/to/picroft-unstable.  The code is also in the Buster branch of this repo, but there really isn't much different from Stretch.

forslund, 2020-01-06:
The buster version has been released, closing this issue.

closed 2020-01-06 by forslund