MycroftAI/enclosure-picroft #115: Trouble setting up Google AIY Voice Kit and solutions.

ConorIA, 2019-08-11:
I'm running Picroft on a Pi 3B. I built it following [_image_recipe.md_](https://github.com/MycroftAI/enclosure-picroft/blob/stretch/image_recipe.md). I ended up pulling down a buster image of Raspbian as it was the image linked to "latest". 

I am using the Google AIY voicehat and I ran into some problems that I thought that I would share. 

1) The Google AIY DKMS module didn't want to build. It turns out there is an issue with the source code that prevents it from building on newer kernels, as outlined here: https://github.com/google/aiyprojects-raspbian/issues/627. There are some [manual changes](https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-514403972) that should fix the issue, but I couldn't get it to work. In the end, I managed to install using a patched _.deb_ file linked to in the [comments](https://github.com/google/aiyprojects-raspbian/issues/627#issuecomment-516635760). Re-running the AIY setup stuff then seemed to work. 

2) The [_AIY-asound.conf_](https://github.com/MycroftAI/enclosure-picroft/blob/stretch/home/pi/AIY-asound.conf) file wasn't pulled in as part of the Mycroft install (I guess I pulled master). I copied it over manually, but was running into an error: `amixer: Unable to find simple control 'Master',0`. I managed to fix this by replacing the first line with `options snd_rpi_googlemihat_soundcard index=0`, which is slightly different from the code in _AIY-asound.conf_. ([source](https://www.raspberrypi.org/forums/viewtopic.php?p=1336717&sid=44e99b2e6518fc68d4b1a5ab539f5f6e#p1336717))

And finally, this isn't specific to Picroft, but everytime I re-install picroft I have trouble with raspotify, so, in the interest of future me, I wonder whether the following can somehow be implemented: 

1) add the following to _/home/pi/.config/pulse/default.pa_.
```
load-module module-native-protocol-tcp auth-ip-acl=127.0.0.1 auth-anonymous=1
```

2) add the line `default-server = 127.0.0.1
` to _/var/cache/raspotify/.config/pulse/client.conf_.

If there is a more elegant way to address the above, I am open to it. I would be interested to know how Spotify is implemented on the Mark I.

Anyway, I am not sure if any of the above is known and has been fixed outside of master, but I thought I would share my experience FWIW.

andlo, 2019-08-24:
I just got a Pi4 and were thinking on trying this with Mycroft. First look into making a Picroft image based on Bustner and not Stretch, and then looking into getting the Google Hoicekit working on that.
Glad I am not strating from scratch and that you have something cooking alreddy :)

andlo, 2019-08-24:
I didnt have any trouble or issuet setting up the AIY on picroft on bustner...Well the asound.conf were missing and thre were some errors durring install, but after it booted it worked….
So I lookinto the errors and hopefull soon gets a image that should install picroft and the AIY.

andlo, 2019-08-25:
well some troubles…..The audioservice hangs after a short while. Have to investegate. It did that on the stretch to in "old days" but cant remember what were the issue.

andlo, 2019-08-28:
Solved the problem ith audioservice hanging just by updating raspbian. 
```
sudo apt-get update 
sudo apt-get upgrade
```
So my picroft image based on bustner is working fine with the voicekit (v1)

penrods, 2019-11-13:
Try the latest https://mycroft.ai/to/picroft-unstable image.  I think the setup is clean now and works with Pi 3/3+/4 and with the AIY hardware.

ServError, 2019-12-02:
It seems that some of the common issues encountered with the AIY installation have been addressed in a new release as per https://github.com/google/aiyprojects-raspbian/issues/643

krisgesling, 2019-12-23:
I just heard from a user that they needed to install @andlo's AIY Hat Skill or else any speech from the device would be cut off after a few seconds. Is that other people's experience currently?

ConorIA, 2019-12-23:
I was experiencing that when I first opened this issue. The Pi I was using has since been repurposed, so I can't speak to current function. It might be related to the audio service hang describes by @andlo, above.

jannewmarch, 2019-12-28:
I just tried installing raspbian-buster_Picroft_2019-11-10 (Stable 19.8.5) and it has the problem referenced by ServError (google/aiyprojects-raspbian#643): aiy-dkms won't build on later Linux kernel. Seems the problem might be solved in aiyprojects but hasn't found its way into the Picroft image. BTW, the unstable image as at Dec 28 is the same as the stable one so no point trying that at present!

ServError, 2020-03-29:
It's pretty straightforward to manually use the new repo until the fix makes its way in. Flash the card, put it in your Pi and either connect ethernet or use the wizard to set up wifi, then connect over SSH (or if you really want, do this over nano on the pi), and modify auto_run.sh. 

Enable the code at line 343 by removing the #

Replace lines 332 and 333 with the following:

`
echo "deb https://packages.cloud.google.com/apt aiyprojects-stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt-get update
`

and save it and reboot the board. Once you're up and running, finish the wizard and then manually install the Google AIY kit skill if necessary. Everything should work.

ServError, 2020-04-15:
@andlo I see the new repo has made its way in to the build via your commit. Could you re-enable aiy-python-wheels as well since it's been fixed?

andlo, 2020-04-16:
@ServError I will do that as soon I can. You could however also if you like make a PR your self. It is always exciting to make PR's to a project. One of my first PR's were actualy adding support for AIY on this project.

ServError, 2020-04-17:
@andlo hah, fair enough, I figured a one line change wasn't worth a PR, but I'll put it in.

ConorIA, 2021-10-30:
I am closing this issue as it has been a long time since I wrote it and I think that all of the issues / tips in this issue are ancient history. The only exception is that I still seem to require the workaround to use pulseaudio TCP on localhost in order to have sound from raspotify. Is anyone else using raspotify in picroft and have a better solution? @forslund maybe?

closed 2021-10-30 by ConorIA