MycroftAI/enclosure-picroft #120: `jq` error during setup wizard

alexbarcelo, 2019-10-10:
I am running a Raspberry Pi 3 with the stable (July IIRC) Picroft image.

I was configuring everything through ssh. I chose the USB configuration, I rebooted the Raspberry ("R" in the next step, during the adjust volume).

After this, the save_choices method keeps failing with a `jq` error:

```
jq - commandline JSON processor [version 1.5-1-a5b5cbe]
Usage: jq [options] <jq filter> [file...]

        jq is a tool for processing JSON inputs, applying the
        given filter to its JSON text inputs and producing the
        filter's results as JSON on standard output.
        The simplest filter is ., which is the identity filter,
        copying jq's input to its output unmodified (except for
        formatting).
        For more advanced filters see the jq(1) manpage ("man jq")
        and/or https://stedolan.github.io/jq

        Some of the options include:
         -c             compact instead of pretty-printed output;
         -n             use `null` as the single input value;
         -e             set the exit status code based on the output;
         -s             read (slurp) all inputs into an array; apply filter to it;
         -r             output raw strings, not JSON texts;
         -R             read raw strings, not JSON texts;
         -C             colorize JSON;
         -M             monochrome (don't colorize JSON);
         -S             sort keys of objects on output;
         --tab  use tabs for indentation;
         --arg a v      set variable $a to value <v>;
         --argjson a v  set variable $a to JSON value <v>;
         --slurpfile a f        set variable $a to an array of JSON texts read from <f>;
        See the manpage for more options.
```

Adding some prints here and there, I see something fishy in the save_choices function. Is as if the setup_stage variable was empty and thus there was some hiccup in the `jq` command for the stage-related variable.

I haven't been able to configure audio in the Raspberry so far, I don't know if it is related.

Although the symptoms are similar, I see no relation with #109.

krisgesling, 2019-10-11:
Hi Alex,
It sounds like it may be related to #116 which has been pushed in the latest [Picroft Unstable image](https://mycroft.ai/to/picroft-unstable). I haven't yet heard of anyone having problems with this image and it's been out for a few weeks so will likely make it the stable image shortly.

Do you want to try that image, and see if you still run into the same problem?

alexbarcelo, 2019-10-11:
@krisgesling Yeah, that sounds like it. Sorry I missed that completely, and don't know why. Maybe dumb-past-me was looking for issues and not PR.

I'll try the unstable image, but the symptoms  *and* causes seem to match perfectly. Thanks for fixing it in that PR!

closed 2019-10-11 by alexbarcelo