MycroftAI/enclosure-picroft #121: Create Picroft Enclosure

krisgesling, 2019-11-11:
Unlike the formal devices, Picroft doesn't currently have an "enclosure" to listen and act upon relevant messages on the messagebus. For example, asking Mycroft to shutdown triggers the Stop Skill, and emits a message, but there is no enclosure to act upon this message.

For simplicity, this could be implemented in a Picroft-Enclosure Skill similar to that currently used by the [Mark 2](https://github.com/MycroftAI/skill-mark-2/blob/master/__init__.py).

Messages to handle would include:
- `system.shutdown`
- `system.reboot`

penrods, 2019-11-12:
I second (or third) this!  See the discussion here:
https://chat.mycroft.ai/community/pl/z3mmzh15x78y8xndorhzyekb1w

penrods, 2019-11-13:
See:  https://github.com/MycroftAI/enclosure-picroft/pull/122

Comments welcome!

krisgesling, 2019-11-16:
Love your work Mr Penrod!

I don't have any Pi's on me at the moment to test, but it looks pretty tidy