MycroftAI/enclosure-picroft #123: updating & adding ui to picroft broke it

jshep321, 2019-11-22:
Hi,
-Running the latest version of picroft (19.8.3) on my RPI3b+...  but I wanted a GUI.
-Running a standard wake word (Jarvis)
-Running custom USB mic and I2S speakers (both working fine before changes below) 

Procedure:
sudo apt update && sudo apt-get upgrade -y
sudo reboot
_-->tested picroft and everything seemed to work OK_

sudo apt install raspberrypi-ui-mods
sudo reboot

then enabled VNC (which automatically installs realvnc server as seen below) and enabled boot to desktop / GUI mode (with password) via raspi-config.

Problems:
1. Had to re-pair the device. 
2. It broke my audio, but the mic test and playback both still work?!?.
3. Some unknown connection errors occur (see below log file)

Resolution attempts:
-I tried resetting the boot mode to console and disabling VNC, but same results. 
-Looking to audio fixes.

Is there something inherent in an update or adding a GUI that should behave like this?

**Below is a partial log:**
 15:54:55.431 | ERROR    |   687 | mycroft.skills.settings:_issue_api_call:311 | Failed to upload skill settings meta for fallback-unknown|19.08
Traceback (most recent call last):
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 343, in _make_request
    self._validate_conn(conn)
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 839, in _validate_conn
    conn.connect()
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connection.py", line 301, in connect
    conn = self._new_conn()
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connection.py", line 164, in _new_conn
    (self.host, self.timeout))
urllib3.exceptions.ConnectTimeoutError: (<urllib3.connection.VerifiedHTTPSConnection object at 0x673541f0>, 'Connection to api.mycroft.ai timed out. (connect timeout=3.05)')

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/home/pi/mycroft-core/.venv/lib/python3.5/site-packages/urllib3/util/retry.py", line 399, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
~~~~-fb2de00bf6f8/settingsMeta (Caused by ConnectTimeoutError(<urllib3.connection.VerifiedHTTPSConnection object at 0x673541f0>, 'Connection to api.mycroft.ai timed out. (connect timeout=3.05)'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/mycroft-core/mycroft/skills/settings.py", line 308, in _issue_api_call
    self.api.upload_skill_metadata(self.settings_meta)
  File "/home/pi/mycroft-core/mycroft/api/__init__.py", line 382, in upload_skill_metadata
    "json": settings_meta
  File "/home/pi/mycroft-core/mycroft/api/__init__.py", line 69, in request
[... 129 lines cut ...]
(1.2.5-1+rpi6, automatic), liblzo2-2:armhf (2.08-1.2, automatic), libhtml-format-perl:armhf (2.12-1, automatic), libevdev2:armhf (1.5.6+dfsg-1, automatic)
End-Date: 2019-11-22  13:40:17

Start-Date: 2019-11-22  13:42:17
Commandline: apt-get install realvnc-vnc-server
Requested-By: pi (1000)
Install: libsane-extras:armhf (1.0.22.4, automatic), libdjvulibre-text:armhf (3.5.27.1-7, automatic), libqpdf17:armhf (6.0.0-2, automatic), liblouisutdml-data:armhf (2.6.0-3, automatic), libjxr0:
armhf (1.1-6, automatic), poppler-utils:armhf (0.48.0-2+deb9u2, automatic), libgusb2:armhf (0.2.9-1, automatic), libieee1284-3:armhf (0.2.11-13, automatic), netpbm:armhf (2:10.0-15.3, automatic),
 bc:armhf (1.06.95-9, automatic), sane-utils:armhf (1.0.25-4.1, automatic), imagemagick:armhf (8:6.9.7.4+dfsg-11+deb9u7, automatic), acl:armhf (2.2.52-3, automatic), libgs9:armhf (9.26a~dfsg-0+de
b9u5, automatic), libmagickwand-6.q16-3:armhf (8:6.9.7.4+dfsg-11+deb9u7, automatic), liblouis12:armhf (3.0.0-3+deb9u4, automatic), fonts-droid-fallback:armhf (1:6.0.1r16-1.1, automatic), cups-ser
ver-common:armhf (2.2.1-8+deb9u4, automatic), libwmf0.2-7:armhf (0.2.8.4-10.6, automatic), cups-common:armhf (2.2.1-8+deb9u4, automatic), ssl-cert:armhf (1.0.39, automatic), colord-data:armhf (1.
3.3-2, automatic), libsane-extras-common:armhf (1.0.22.4, automatic), libpaper1:armhf (1.1.24+nmu5, automatic), liblqr-1-0:armhf (0.4.2-2, automatic), cups-filters:armhf (1.11.6-3+rpt2+deb9u1, au
tomatic), imagemagick-6.q16:armhf (8:6.9.7.4+dfsg-11+deb9u7, automatic), cups-ppdc:armhf (2.2.1-8+deb9u4, automatic), libcupsmime1:armhf (2.2.1-8+deb9u4, automatic), libjbig2dec0:armhf (0.13-4.1,
 automatic), libilmbase12:armhf (2.2.0-12, automatic), liblouisutdml7:armhf (2.6.0-3+b1, automatic), libcupsfilters1:armhf (1.11.6-3+rpt2+deb9u1, automatic), printer-driver-gutenprint:armhf (5.2.
11-1+b6, automatic), realvnc-vnc-server:armhf (6.6.0.41949), libpaper-utils:armhf (1.1.24+nmu5, automatic), ghostscript:armhf (9.26a~dfsg-0+deb9u5, automatic), libfontembed1:armhf (1.11.6-3+rpt2+
deb9u1, automatic), libgutenprint2:armhf (5.2.11-1+b6, automatic), libijs-0.35:armhf (0.35-12, automatic), libfile-copy-recursive-perl:armhf (0.38-1, automatic), libopenexr22:armhf (2.2.0-11, aut
omatic), libmagickcore-6.q16-3:armhf (8:6.9.7.4+dfsg-11+deb9u7, automatic), liblouisutdml-bin:armhf (2.6.0-3+b1, automatic), libgs9-common:armhf (9.26a~dfsg-0+deb9u5, automatic), gsfonts:armhf (1
:8.11+urwcyr1.0.7~pre44-4.3, automatic), qpdf:armhf (6.0.0-2, automatic), cups-filters-core-drivers:armhf (1.11.6-3+rpt2+deb9u1, automatic), libdjvulibre21:armhf (3.5.27.1-7, automatic), libsane-
common:armhf (1.0.25-4.1, automatic), libcupsppdc1:armhf (2.2.1-8+deb9u4, automatic), liblouis-data:armhf (3.0.0-3+deb9u4, automatic), libmagickcore-6.q16-3-extra:armhf (8:6.9.7.4+dfsg-11+deb9u7,
 automatic), libjxr-tools:armhf (1.1-6, automatic), libnetpbm10:armhf (2:10.0-15.3, automatic), cups-core-drivers:armhf (2.2.1-8+deb9u4, automatic), cups-daemon:armhf (2.2.1-8+deb9u4, automatic),
 libyaml-0-2:armhf (0.1.7-2, automatic), libcolorhug2:armhf (1.3.3-2, automatic), libcupsimage2:armhf (2.2.1-8+deb9u4, automatic), libpoppler64:armhf (0.48.0-2+deb9u2, automatic), fonts-noto-mono
:armhf (20161116-1, automatic), cups:armhf (2.2.1-8+deb9u4, automatic), libcupscgi1:armhf (2.2.1-8+deb9u4, automatic), poppler-data:armhf (0.4.7-8, automatic), cups-client:armhf (2.2.1-8+deb9u4,
automatic), libpng12-0:armhf (1.2.54-6, automatic), libsane:armhf (1.0.25-4.1, automatic), update-inetd:armhf (4.44, automatic), cups-browsed:armhf (1.11.6-3+rpt2+deb9u1, automatic), colord:armhf
 (1.3.3-2, automatic), imagemagick-6-common:armhf (8:6.9.7.4+dfsg-11+deb9u7, automatic)
End-Date: 2019-11-22  13:45:12

penrods, 2019-11-26:
Basically -- don't do that.  :)

There is a reason we are using the "lite" versions of Raspbian.  The GUI adds processor load and complications.  Mycroft on a Pi 2 with GUI was impossible, and on the Pi 3/3+ is was possible but basically unusable.  

What it boils down to is that a Pi has just enough power to be a good dedicated device as a voice agent.  A desktop operating environment adds the overhead of the desktop itself (X windows plus the desktop) PLUS you are going to be running even more applications on top of that.

The Pi 4 is getting close to having enough power to run both a desktop with minimal applications.  We will likely be offering a dedicated GUI client that can run on the Pi4, but that is not the same as running a full desktop OS.

The Pi 4 is great, but it isn't even as powerful as a cheap laptop.

penrods, 2019-11-26:
I should add that for your immediate purposes it is probably easier to re-flash a fresh Picroft image and start from scratch.  Un-doing is often much harder than re-doing, and I think that is the case here.  I'd keep the old image on an independent SD card until you get it working, just in case you need to copy anything you might need from it during the recovery.

closed 2019-11-26 by penrods

jshep321, 2019-11-26:
Ok, ok, I undid it by starting over again as identifying the conflicts was a time suck.  

Although I am a mere beginner at mycroft,  I normally like to know the design challenges -- functionality (/compatibility) vs performance.  It seems that something went awry other than mere CPU horsepower here.  The pi's pixel desktop is very lean compared to other desktops/window managers, especially when it's simply idle.  

Also, I have a pretty good idea of the CPU requirements of chromium in kiosk mode and it will almost certainly run in the overhead of picroft once its stable on the pi4 from a CPU perspective.  Since the RAM is (up to) 4X in the pi4, that should not be a restriction either.

Perhaps a path forward is to run mycroft in a container on the pi4 in the future... but I'm out of tinkering time for this unfortunately.