MycroftAI/enclosure-picroft #133: Remove PS3 Eye support

StuartIanNaylor, 2020-03-26:
Until a fix is found for the PS3 Eye its probably best not to recommend it.
Doubt it though as this has a history over a decade long.
alsactl init uses a generic usbmixer rules file and the ps3 eye was a very specific device for the ps3.
https://www.systutorials.com/docs/linux/man/7-alsactl_init/

Both Udev & alsactl init get some of the control and info registers wrong.
You will often see in the Journal lines such as 
```
kernel: usb 1-1.1.1: 3:1: cannot get min/max values for control 2 (id 3)
```

as `alsactl -d init` get things wrong with a default generic usb-mixer rule.
Also the read register of current bit rate is obviously wrong and seems to be pointing at some dynamic value that is highly likely to have nothing to do with current sample rate.

These errors go back a decade and the internet is littered with them but I seem to be unable to get a single fix.
Its very likely a rule could be defined and included in /usr/share/alsa/init/00main but exactly what I don't know.

Alsa detects enough to get generic sampling working but has never completely configured the Sony implementation.
https://jim.sh/svn/jim/devl/playstation/ps3/eye/test/eye.c
https://www.systutorials.com/docs/linux/man/7-alsactl_init/
```
pi@raspberrypi:~ $ alsactl -d init 1
alsactl: sysfs_init:43: sysfs_path='/sys'
alsactl: new_root_dir:1276: new_root_dir '/usr/share/alsa/init/00main' '/usr/share/alsa/init'
alsactl: parse:1665: start of file '/usr/share/alsa/init/00main'
alsactl: parse:1725: read (5) 'CONFIG{sysfs_device}="/class/sound/card$cardinfo{card}/device"'
alsactl: get_key:897: operator=assign
alsactl: get_key:907: key='CONFIG{sysfs_device}'
alsactl: get_key:933: value='/class/sound/card$cardinfo{card}/device'
alsactl: get_key_attribute:958: attribute='sysfs_device'
alsactl: apply_format:1059: will substitute format name 'cardinfo'
alsactl: get_format_attribute:980: attribute='card', str='/device'
alsactl: apply_format:1091: format=1, string='/class/sound/card', tail='/device'
alsactl: apply_format:1102: substitute cardinfo{card} '1'
alsactl: parse_line:1635: CONFIG{sysfs_device}='/class/sound/card1/device'
alsactl: parse:1725: read (6) 'ACCESS!="$sysfsroot$config{sysfs_device}",
        CONFIG{sysfs_device}="/class/sound/controlC$cardinfo{card}/device"'
alsactl: get_key:889: operator=nomatch
alsactl: get_key:907: key='ACCESS'
alsactl: get_key:933: value='$sysfsroot$config{sysfs_device}'
alsactl: apply_format:1059: will substitute format name 'sysfsroot'
alsactl: apply_format:1091: format=5, string='', tail='$config{sysfs_device}'
alsactl: apply_format:1180: substitute sysfs_path '/sys'
alsactl: apply_format:1059: will substitute format name 'config'
alsactl: get_format_attribute:980: attribute='sysfs_device', str=''
alsactl: apply_format:1091: format=7, string='/sys', tail=''
alsactl: parse_line:1581: access(/sys/class/sound/card1/device) = 0 ($sysfsroot$config{sysfs_device})
alsactl: parse:1725: read (10) 'ENV{CMD}=="help", INCLUDE="help", GOTO="00main_end"'
alsactl: get_key:885: operator=match
alsactl: get_key:907: key='ENV{CMD}'
alsactl: get_key:933: value='help'
alsactl: get_key_attribute:958: attribute='CMD'
alsactl: parse_line:1458: env: 'CMD' '(null)'
alsactl: parse:1725: read (11) 'ENV{CMD}=="info", INCLUDE="info", GOTO="00main_end"'
alsactl: get_key:885: operator=match
alsactl: get_key:907: key='ENV{CMD}'
alsactl: get_key:933: value='info'
alsactl: get_key_attribute:958: attribute='CMD'
alsactl: parse_line:1458: env: 'CMD' '(null)'
alsactl: parse:1725: read (12) 'ENV{CMD}=="default", INCLUDE="default", GOTO="00main_end"'
alsactl: get_key:885: operator=match
alsactl: get_key:907: key='ENV{CMD}'
alsactl: get_key:933: value='default'
alsactl: get_key_attribute:958: attribute='CMD'
alsactl: parse_line:1458: env: 'CMD' '(null)'
alsactl: parse:1725: read (13) 'ENV{CMD}=="test", INCLUDE="test", GOTO="00main_end"'
alsactl: get_key:885: operator=match
alsactl: get_key:907: key='ENV{CMD}'
alsactl: get_key:933: value='test'
alsactl: get_key_attribute:958: attribute='CMD'
[... 1559 lines cut ...]
alsactl: get_key:933: value='1'
alsactl: get_key_attribute:958: attribute='do_search'
alsactl: parse_line:1354: ctl match: 'do_search' '1' '0'
alsactl: do_match:494: match CTL{do_search} '1' <-> '0'
alsactl: do_match:504: CTL{do_search} is false
alsactl: parse:1725: read (284) 'CTL{name}="Internal Mic Boost",CTL{do_search}=="1", CTL{values}="on"'
alsactl: get_key:897: operator=assign
alsactl: get_key:907: key='CTL{name}'
alsactl: get_key:933: value='Internal Mic Boost'
alsactl: get_key_attribute:958: attribute='name'
alsactl: parse_line:1331: ctl assign: 'Internal Mic Boost' 'name'
alsactl: get_key:885: operator=match
alsactl: get_key:907: key='CTL{do_search}'
alsactl: get_key:933: value='1'
alsactl: get_key_attribute:958: attribute='do_search'
alsactl: parse_line:1354: ctl match: 'do_search' '1' '0'
alsactl: do_match:494: match CTL{do_search} '1' <-> '0'
alsactl: do_match:504: CTL{do_search} is false
alsactl: parse:1739: end of file '/usr/share/alsa/init/default'
alsactl: parse:1725: read (45) 'EXIT="99"'
alsactl: get_key:897: operator=assign
alsactl: get_key:907: key='EXIT'
alsactl: get_key:933: value='99'
alsactl: parse:1739: end of file '/usr/share/alsa/init/00main'
```

krisgesling, 2020-03-30:
Hey Stuart, 

Just wanted to post a brief copy of the chat from last week, mostly for the benefit of anyone stumbling across this in the future. 

I think the less we have to implement custom fixes and stick to defaults the better. However we aren't going to remove PS3 Eye support because we have significant number of people who use this hardware as their primary configuration. 

We do very much appreciate the amount of work you've been putting in to investigate other solutions though. I hope we can find a path forward that provides a clean and intuitive audio setup, without breaking core functionality for our PS Eye users.

StuartIanNaylor, 2020-03-30:
Afraid ground to a halt I am afraid.

> Hi Stuart,
> 
> I haven't used this hardware in quite a while at this point.  Do you
> have the same problem on desktop, or just the Pi?  You might try
> disabling USB autosuspend (on the pi, add "usbcore.autosuspend=-1" to
> /boot/cmdline.txt).  But that's just a guess.
> 
> If you have a kernel version that consistently works and one that
> consistently does not, a "git bisect" would be the best way to track
> it down: https://www.kernel.org/doc/html/latest/admin-guide/bug-bisect.html
> If the bug only shows up sometimes, though, that would be hard to do.
> 
> Jim

I really would suggest they don't source a PS3 eye if you don't already have one.

For a few dollar more you can get a respeaker 2 mic for the pi that has a few Leds and a small amp that allows you to create a complete simple device AI.

That would be my recommendation.

StuartIanNaylor, 2020-04-06:
I keep having a go at the PS3 eye and we don't even have alsa asoundrc files.

This will set a default took me a while to work out the simple async but hey!

```
pcm.array {
 type hw
 card 1
}

pcm.array_gain {
 type softvol
   slave {
   pcm "array"
   }
 control {
   name "Mic Gain"
   count 2
   card 1
   }
 min_dB -40.0
 max_dB 10.0
 resolution 80
}

pcm.cap {
 type plug
 slave {
   pcm "array_gain"
   channels 4
   }
 route_policy sum
}

pcm.!default {
    type asym
    playback.pcm "plughw:CARD=ALSA,DEV=0"
    capture.pcm  "cap"
}
```
The default will be cap which is the 4 channels addded together.
The other 2 are available if that pcm is designated.
Gave up on the softvolume what should work doesn't seem to and prob due to the underlying control error of the ps3 eye
Also because plughw is used rather than hardware direct it will get software conversion