MycroftAI/enclosure-picroft #135: Incorrect debian repo for Google AIY

tbertz, 2020-03-29:
Setup: Raspberry Pi 3, Google AIY Mic and Speaker Hat

Steps: Create image from latest release, run on Pi, attempt guided install, get to Audio choices, select Google AIY

Result: Install of deb repo's fails with messages about gpg signatures

I think the cause of this is that you are using old references for the repo and key in enclosure-picroft and auto_run.sh on line 332 and 333

echo "deb https://dl.google.com/aiyprojects/deb stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list
wget -q -O - https://dl.google.com/linux/linux_signing_key.pub | sudo apt-key add -

The up to date references according to Google own documentation (https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) are;

Add AIY package repo:

echo "deb https://packages.cloud.google.com/apt aiyprojects-stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list

Add Google package keys:

curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -

Installs ran fine following my manual change to file on device. However I then got stuck at the rainbow boot screen. But I'll file a separate issue for this once I've tried a fresh install with this issue resolved before reporting.

andlo, 2020-03-29:
Hi @tbertz 
thank you for your participation in the project. I have made correction as you have so well described and made a PR.

tbertz, 2020-03-29:
No problems, appreciate the hard work. Hope I can get it up and successfully running.

krisgesling, 2020-03-31:
Thanks for the help tbertz!

If you want to be added to [our list of excellent human beings](https://github.com/MycroftAI/contributors) please complete a Contributor Licensing Agreement by entering your details at https://mycroft.ai/cla but totally up to you.

closed 2020-03-31 by krisgesling

tbertz, 2020-04-10:
Happy to. Unfortunately get a database connection error when I submit my details on the page you provided the link to.

krisgesling, 2020-04-10:
Hey thanks for flagging that error, I've fixed that up now.

krisgesling, 2020-04-13:
You can also [email me](mailto:kris.gesling@mycroft.ai) your name, and confirm you're contributing as an individual (not on company time or through an employer), and I'll kick off the signing process.