MycroftAI/enclosure-picroft #137: Provide torrent for image?

tobymurray, 2020-04-05:
I am just looking at getting started Mycroft AI, and Picroft seems like the cheapest/easiest way to get into it. One of the main drivers for being interested in Mycroft AI is I have a terrible internet connection, and ultimately I'd hope to be able to run this offline.

The current picroft image (retrieved from [here](https://mycroft.ai/to/picroft-image)) is ~2.6GB. Trying to download this on a shoddy internet connection is a mess, as it both downloads slowly and is not stable enough to survive the ~12 - 14 hours required. 

Any way you could host a torrent (or similar) for the image? Even with a single seeder, it's dramatically more accessible.

krisgesling, 2020-04-06:
Hey Toby, thanks for the good suggestion, it's been floated as an idea previously but not something we've looked into in detail. 

I've flagged this as `help wanted` to see if anyone is familiar with the end-to-end process and can suggest one or more reliable trackers to host it through. It's certainly something we could explore as part of our release process.

im-strongthany, 2020-04-07:
I had previously worked on setting one up, but life changes had made the project go to the wayside. I am currently not able to work on it on my Mark 1, but I'll see if I am able to get to work on it again but expect delays because of personal life changes again and the ever present COVID. 

I'll update this in a week with whatever updates I have. Set your expectations for a torrent seed, not a skill(yet).

Thanks!

im-strongthany, 2020-04-14:
Hey @tobymurray, I said I'd have an update and here I am. Like I guessed I don't have a skill but I do have a torrent of the image. I have posted it on the [Mycroft Chat](https://chat.mycroft.ai), and am linking to that download [here](https://chat.mycroft.ai/files/w9wai8k9tf893q379ktybmpuke/public?h=M39HlZBGs70YXfbWC08nhsr-WgrIsWtgNv31UDbKv0I). As far as I know I am the only one seeding this, but I'll try to do some outreach to see if I can get some more people on this. Let me know if you run into any issues. Thanks!

krisgesling, 2020-04-15:
Hey @im-strongthany, thanks for pushing that out. 

If we're going to do this long-term it would be best if we can get a process setup to generate it automatically on each release.

Are you able to write up a quick outline of your process and some tracker recommendations?

tobymurray, 2020-04-16:
I'll try downloading it today, see how it goes. I have "up to 1 Mbps" upstream, so I won't be a very effective seeder, but I can leave it up.

fermulator, 2020-06-21:
@im-strongthany-zz if you can document how you did it, I can add it to the automation process, 
(also added the torrent myself to my seeder box for now)

MKE1138, 2021-03-11:
Hi all, I've got a seedbox that I can lend to this, and I'd also be willing to host a small & inexpensive VPS to just act like a torrent seeder.

krisgesling, 2021-03-16:
In the interim, we've moved the download to a dedicated file server that will hopefully provide better download speeds.

Same link: https://mycroft.ai/to/picroft-image