MycroftAI/enclosure-picroft #139: Community Crowd Source of Information for Thermal Management?

fermulator, 2020-06-21:
Hey team,

Please take a gander at my initial uncertainty after a first-time transition of MyCroft onto a raspberry pi.
https://www.reddit.com/r/Mycroftai/comments/hcr59z/mycroft_ai_picroft_running_the_pi_too_hot/

Would we be open to starting a document that allows the community to contribute and share their mycroft performance and thermal management?

Was thinking ... similar to "supported hardware", maybe we add a table that includes:
* Date Sampled
* Raspberry Pi HW Version
* enclosure-picroft Version
* mycroft-core Version
* Pi HW Thermal Management (heatsinks Y/N etc w/ links)
* Pi HW Case (which case w/ links)
* CPU Temp
* GPU Temp
* Load Average / Idle %

The GOAL here, is to provide some recommendations in terms of pi HW enclosure/case, including most important thermal management expectations.

PS: perhaps github docs is too difficult ; if there is a different / easier way to to maintain that'd be better (wiki? maybe I write a script you could run that outputs MD table format with everything except the case/HW user filles that in?)

krisgesling, 2020-07-01:
Hey, if you would find it useful then I say go for it!

I wonder if a basic Google Sheet would be easiest?

A script to collect the data would be good, much better chance of consistent data.

BenjaminHCCarr, 2020-07-01:
@fermulator there is a WIKI here
https://github.com/MycroftAI/enclosure-picroft/wiki

Would be nice if there was a way to automate submitting
I have a Mycroft Pi3B in an aluminum enclosure but  it isn't running at the moment

fermulator, 2020-07-03:
1. Started: https://github.com/MycroftAI/enclosure-picroft/wiki/User-Community-Pi-Samples
2. PoC Script ready: #140

fermulator, 2020-10-31:
Poked https://chat.mycroft.ai/community/pl/3ak81hn5wpbidbrtiukmyoqwqo to figure out if we can move this forward.

fermulator, 2022-04-05:
(assuming idea is dead)

closed 2022-04-05 by fermulator