MycroftAI/enclosure-picroft #142: Is picroft supported on Pi4 with 8GB

robertsawko, 2020-08-08:
Hello,

I have only just started playing with Picroft and I burned the Picroft-Buster-Keaton-2020-01-10 stable on an SD card. After connecting all the components and turning on the power supply, I get stack on this bootloader screen

```
Raspberry Pi 4 - 8GB
bootloadr: 94of978d Mar 19 2020

board: d03114 2ee894c0 dc:a6:32:b4:4e:4f
boot: mode 6 order 0x00000001 rsts 0x00001000
SC CID: 00035344534c333247808b4eb3590136
part: 0 mbr [0x0c:00002000 0x83:00082000 0x00:00000000 0x00:00000000]
fw: start4.elf fixup4.dat
net down ip: 0.0.0.0 sn:0.0.0 gw: 0.0.0
tftp 0.0.0.0

Partition: 0
lba 8192 oem: 'mkfs.fat' volume: ' boot        '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2-rsec 0
Read config.txt bytes      1784 hnd 0x000002d3 hash 6c0f63e7068a74e6
recovery4.elf not found
recovery.elf not found
Read start4.elf bytes 2769540 hnd 0x0000f80a hash 7d1a9397d2af5348
Read fixup4.dat bytes    6167 hnd 0x000002e5 hash 4e9d3ac543571c66
0x00d03114 0x00000000 0x00000000
start4.elf: is not compatible 0x0
This board requires newer software. Get the latest software from:-
     https://www.raspberrypi.org/downloads
```
I am guessing that the last statement says it all, but I just wanted to verify with you. The Pi4 is marked as supported on your documentation pages(?), but perhaps that wasn't adjusted for the latest model?

Please let me know.

krisgesling, 2020-08-10:
Thanks Robert, we haven't pushed a new Picroft image in a while, so I presume this will work fine with an updated image though I don't have an 8GB model to test it on.

We're about to do a major release of core (v20.8) so will likely hold off until that is out at the end of this month.

moelha, 2020-08-12:
I have not connected it to an external screen, but I do get a similar issue. I have RPi 4 with 8GB of RAM, and it does not boot. I get "Unsupported board type" error (4 long flashes, 4 short flashes). I have also tried the official Raspbian OS (2020-05-27) lite, which works just fine. I guess the current Picroft image is not compatible with the RPi 4 (at least with the 8GB version) as it's based on an older version of the Raspbian OS.

krisgesling, 2021-06-03:
Hi all, the Picroft v20.08 images and above should all work on an 8GB Pi 4 now.

closed 2021-06-03 by krisgesling

robertsawko, 2021-06-07:
Thanks for all the comments and updates.