MycroftAI/enclosure-picroft #146: Multiple Issues on Buster PiCroft

binkocd, 2020-10-18:
## Hardware

Raspberry Pi 3, Google AIY v1

## Replicate the Issue

1. Burn image to Micro SD card using Etcher
2. Seat the Micro SD card in the RPi 3
3. Boot Picroft
4. Wait 3 minutes
5. SSH into Picroft
6. Run through Wizard
7. In hardware setup, select '4' for Google AIY
8. Get first error for Google AIY Project 
`W: GPG error: https://dl.google.com/aiyprojects/deb stable Release: The following signatures were invalid: EXPKEYSIG 6494C6D6997C215E Google Inc. (Linux Packages Signing Authority) <linux-packages-keymaster@google.com>
E: The repository 'https://dl.google.com/aiyprojects/deb stable Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package aiy-dkms
E: Unable to locate package aiy-io-mcu-firmware
E: Unable to locate package aiy-vision-firmware
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package aiy-dkms
E: Unable to locate package aiy-voicebonnet-soundcard-dkms
E: Unable to locate package aiy-voicebonnet-routes
`
9. Allow Picroft setup to reboot.
10. SSH back into picroft and receive following error:
`Last login: Sun Oct 18 04:25:20 2020

 ███╗   ███╗██╗   ██╗ ██████╗██████╗  ██████╗ ███████╗████████╗
 ████╗ ████║╚██╗ ██╔╝██╔════╝██╔══██╗██╔═══██╗██╔════╝╚══██╔══╝
 ██╔████╔██║ ╚████╔╝ ██║     ██████╔╝██║   ██║█████╗     ██║   
 ██║╚██╔╝██║  ╚██╔╝  ██║     ██╔══██╗██║   ██║██╔══╝     ██║   
 ██║ ╚═╝ ██║   ██║   ╚██████╗██║  ██║╚██████╔╝██║        ██║   
 ╚═╝     ╚═╝   ╚═╝    ╚═════╝╚═╝  ╚═╝ ╚═════╝ ╚═╝        ╚═╝   

        _____    _                          __   _   
       |  __ \  (_)                        / _| | |  
       | |__) |  _    ___   _ __    ___   | |_  | |_ 
       |  ___/  | |  / __| | '__|  / _ \  |  _| | __|
       | |      | | | (__  | |    | (_) | | |   | |_ 
       |_|      |_|  \___| |_|     \___/  |_|    \__|

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/mycroft-core/mycroft/__init__.py", line 17, in <module>
    from mycroft.api import Api
  File "/home/pi/mycroft-core/mycroft/api/__init__.py", line 22, in <module>
    from mycroft.configuration import Configuration
  File "/home/pi/mycroft-core/mycroft/configuration/__init__.py", line 15, in <module>
    from .config import Configuration, LocalConf, RemoteConf
  File "/home/pi/mycroft-core/mycroft/configuration/config.py", line 23, in <module>
    from mycroft.util.json_helper import load_commented_json, merge_dict
  File "/home/pi/mycroft-core/mycroft/util/__init__.py", line 24, in <module>
    from mycroft.util.format import nice_number
  File "/home/pi/mycroft-core/mycroft/util/format.py", line 32, in <module>
    import lingua_franca.format
  File "/home/pi/mycroft-core/.venv/lib/python3.7/site-packages/lingua_franca/format.py", line 20, in <module>
    from lingua_franca.lang.format_en import *
  File "/home/pi/mycroft-core/.venv/lib/python3.7/site-packages/lingua_franca/lang/format_en.py", line 19, in <module>
    from lingua_franca.lang.common_data_en import _NUM_STRING_EN, \
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 857, in get_code
  File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
ValueError: bad marshal data (unknown type code)
`
11. Proceed with testing and get following error:
`Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.7/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/mycroft-core/mycroft/__init__.py", line 17, in <module>
    from mycroft.api import Api
  File "/home/pi/mycroft-core/mycroft/api/__init__.py", line 22, in <module>
    from mycroft.configuration import Configuration
  File "/home/pi/mycroft-core/mycroft/configuration/__init__.py", line 15, in <module>
    from .config import Configuration, LocalConf, RemoteConf
  File "/home/pi/mycroft-core/mycroft/configuration/config.py", line 23, in <module>
    from mycroft.util.json_helper import load_commented_json, merge_dict
  File "/home/pi/mycroft-core/mycroft/util/__init__.py", line 24, in <module>
    from mycroft.util.format import nice_number
  File "/home/pi/mycroft-core/mycroft/util/format.py", line 32, in <module>
    import lingua_franca.format
  File "/home/pi/mycroft-core/.venv/lib/python3.7/site-packages/lingua_franca/format.py", line 20, in <module>
    from lingua_franca.lang.format_en import *
  File "/home/pi/mycroft-core/.venv/lib/python3.7/site-packages/lingua_franca/lang/format_en.py", line 19, in <module>
    from lingua_franca.lang.common_data_en import _NUM_STRING_EN, \
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 857, in get_code
  File "<frozen importlib._bootstrap_external>", line 525, in _compile_bytecode
ValueError: bad marshal data (unknown type code)
`

## Expected condition, and the deviation from expected condition. 

Assume I am supposed to download things from AIY Project, run through project, and have a working Picroft.

## Provide log files or other output to help us see the error

See Above

andlo, 2020-10-18:
Hmmm that is strange - I just did the exactly same and all went smoothly. I did use the unstable version though. 
I also have a google AIY voicekit. 
Have you tryed the unstable downlodad (just to se if yo get same error ?

binkocd, 2020-10-18:
@andlo I used stable. I was looking around in issues, the repo, etc and it looks like it _might_ be solved, though tangentially, but not part of the stable image. I started looking at another issue (will look for it again in a bit when near my dev box, where I have a bunch of notes from last night) and it seemed like there were some easy fixes that could get you going, but not for the faint of heart or those who don't have a lot of time to spend. More of a cobbler's approach. :smiley: 

I would recommend, for us that are just wanting to try it out, who have been interested for a while, but don't have a lot of in-depth project knowledge, making it obvious that the stable image isn't currently working out of the gate. 

I will try to get some time tonight or this week to give the unstable image a shot. Thank you for your quick response, I look forward to giving this a go.

andlo, 2020-10-19:
You could be right - @krisgesling would be the man for adding info about the stable (if something is known not to work) or promote the unstable to stable.

binkocd, 2020-10-20:
I installed the unstable image, still seeing some issues in the output as it's running through the setup wizard:

`HARDWARE SETUP
How do you want Mycroft to output audio:
  1) Speakers via 3.5mm output (aka 'audio jack' or 'headphone jack')
  2) HDMI audio (e.g. a TV or monitor with built-in speakers)
  3) USB audio (e.g. a USB soundcard or USB mic/speaker combo)
  4) Google AIY Voice HAT and microphone board (Voice Kit v1)
  5) ReSpeaker Mic Array v2.0 (speaker plugged in to Mic board)
Choice [1-5]: 4 - Google AIY Voice HAT and microphone board (Voice Kit v1)
deb https://dl.google.com/aiyprojects/deb stable main
OK
Ign:1 https://dl.google.com/aiyprojects/deb stable InRelease
Get:2 https://dl.google.com/aiyprojects/deb stable Release [934 B]
Get:3 https://dl.google.com/aiyprojects/deb stable Release.gpg [819 B]     
Get:4 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]      
Get:5 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Ign:3 https://dl.google.com/aiyprojects/deb stable Release.gpg        
Get:6 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:7 http://archive.raspberrypi.org/debian buster/main armhf Packages [331 kB]
Get:8 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.7 kB]       
Reading package lists... Done                                                                
W: GPG error: https://dl.google.com/aiyprojects/deb stable Release: The following signatures were invalid: EXPKEYSIG 6494C6D6997C215E Google Inc. (Linux Packages Signing Authority) <linux-packages-keymaster@google.com>
E: The repository 'https://dl.google.com/aiyprojects/deb stable Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package aiy-dkms
E: Unable to locate package aiy-io-mcu-firmware
E: Unable to locate package aiy-vision-firmware
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package aiy-dkms
E: Unable to locate package aiy-voicebonnet-soundcard-dkms
E: Unable to locate package aiy-voicebonnet-routes
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package leds-ktd202x-dkms
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard
cp: cannot stat 'AIY-asound.conf': No such file or directory
Installing packages...
`
It seems as though it does install everything and runs through the audio tests fine. So that's some progress. I'll keep chugging along as time allows. 

If @krisgesling has any input on the stable image being unstable vs promoting the unstable vs something else, I would be interested in following up.

krisgesling, 2020-10-20:
Hey, the "unstable" image has been going well, I think this is the first real issue anyone's flagged with it. So I think we can promote it to stable as it's been out for over a month now.

binkocd, 2020-10-22:
@krisgesling Thank you for following up. I'll continue to play around with it and see if there is anything else, but I think for the purposes of this issue, we can close once unstable is promoted. Thank you, I look forward to getting more involved as time allows!

krisgesling, 2020-10-28:
Promoted :tada:

closed 2020-10-28 by krisgesling