MycroftAI/enclosure-picroft #147: Usage of Matrix Voice led ring

thorstenMueller, 2020-10-21:
Hello.

I'm thinking on implementing a feature to use the matrix voice led ring.
Might be as known in other smart voice assistants where the ring shows activity during microphone or tts is active.

Would that be on interest?

1NFR4R3D, 2020-11-26:
I have a 6 mic array from seeedstudio (respeaker) that has LEDs built in as well. I am planning on trying that out over this weekend, and if the mic works, I'd like to help get a more generalised implementation working for the LEDs, making it work for both our devices.

thorstenMueller, 2020-11-26:
Honestly i've not started with this project due the lack of spare time. Before you begin maybe it's worth a look at this thread: https://community.mycroft.ai/t/respeaker-4-mic-array-hat-mycroft-a-i-skill/5773

1NFR4R3D, 2020-11-28:
Thanks for the heads up!