MycroftAI/enclosure-picroft #151: Google AIY v1 with Mycroft 20.08-no mic or top button

nerlins, 2021-01-02:
First, I cannot find any way of downloading and testing prior images to 20.08. I’m not sure if anyone is aware, but if you go to a previous Github release page and click on either the stable or unstable it always redirects to the current Picroft_v20.08_2020-09-07.img.

If I install this image I can run Picroft, but the mic does not work. It acts like it sees it, but the dB level is always stock at 7 or something. Installing the Google AIY voicekit skillset doesn't enable the mic or the top button.

The only way the mic and top button work, are with this hacked image: https://community.mycroft.ai/t/mycroft-with-google-aiy-voice-kit-disk-image/2607 . This is based on 9.14 Mycroft. The image performs a LONG forced update to 20.08 upon reboot, and hangs up after a while.

**Environment**
 - Device type: Raspberry Pi 3B
 - OS: Picroft
 - Mycroft-core version:  20.08

nerlins, 2021-01-02:
After hours of troubleshooting, this might be the fix: https://github.com/google/aiyprojects-raspbian/issues/724
 
The new PulseAudio breaks the AIY drivers. Running the command in that post makes the mic work. I am still making sure everything else works. I wouldn't call this fixed just yet.

aandroide, 2021-01-03:
> After hours of troubleshooting, this might be the fix: [google/aiyprojects-raspbian#724](https://github.com/google/aiyprojects-raspbian/issues/724)
> 
> The new PulseAudio breaks the AIY drivers. Running the command in that post makes the mic work. I am still making sure everything else works. I wouldn't call this fixed just yet.

hello do you use your own raspbian image and custom google assistant scripts?
I confirm that file 91 can be deleted or modified with the code that I link to you in the discussion I had with pulseaudio. https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/issues/1082

aandroide, 2021-01-03:
we also talked about it here: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=294600&p=1783790#p1783790

aandroide, 2021-01-03:
I who use the gassistpi project, uninstalled pulseaudio at the end because it was giving me too many problems, it conflicted with files that use pyaudio and I could not adjust the volume when running systemctl. i went back to alsa but i had to go through this procedure:
sudo apt purge pulseaudio pulseaudio-module-bluetooth
sudo apt-get remove libportaudio2
sudo apt-get install libasound2-dev
git clone -b alsapatch https://github.com/gglockner/portaudio
cd portaudio
./configure && make
sudo make install
sudo ldconfig
cd ..
and now everything is working very well.

nerlins, 2021-01-07:
I believe this was primarily a Debian Buster problem, and not Mycroft. I am closing the issue.

closed 2021-01-07 by nerlins

quinnsam, 2021-01-09:
> 
> 
> I believe this was primarily a Debian Buster problem, and not Mycroft. I am closing the issue.

> After hours of troubleshooting, this might be the fix: [google/aiyprojects-raspbian#724](https://github.com/google/aiyprojects-raspbian/issues/724)
>
> The new PulseAudio breaks the AIY drivers. [...]

I have this exact issue and have been trying to fix it for a while now. I looked at the "google/aiyprojects-raspbian" issue that you referenced but didn't see any command to run. 
Do you mind sharing what steps you took to fix the PulseAudio driver?

aandroide, 2021-01-09:
@quinnsam
 Sudo rm /lib/udev/rules.d/91-pulseaudio-rpi.rules
Reboot and test.

quinnsam, 2021-01-12:
@aandroide 
I started from scratch with a fresh image and after installation the mic was working out of the box. Could have been an upstream fix or something but I didn't even need to run the commands that you provided. But thank you anyway!

aandroide, 2021-01-12:
@quinnsam you just succeeded if you downloaded after the day
 Jan 8 as we seem to have fixed that file.