MycroftAI/enclosure-picroft #152: Training fails on current build

sparky-vision, 2021-05-02:
Running on a Pi 4, clean re-image of Picroft Stable 2020-09-07.

## Try to provide steps that we can use to replicate the Issue

For example: 

1. Burn the 2020-09-07 image to Micro SD card using Etcher
2. Seat the Micro SD card in the RPi 4
3. Boot Picroft
4. Go through the automated setup, choose passwords and USB mics, etc.
5. Ctrl+C out of the CLI into a Linux prompt.
6. CD into "mycroft-precise"
7. Run "source .venv/bin/activate/
8. Add all the training data via SFPT
9. Run "precise-train"
10. Receive the following error: "ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 44 from C header, got 40 from PyObject"

I tried installing various versions of numpy, which led to a variety of errors, each one different than the last.

## Be as specific as possible about the expected condition, and the deviation from expected condition. 

I expect training to proceed. I was apparently running an older version that didn't have issues, I re-imaged because I ran into prettyparse errors, so I tried a clean install.

## Provide log files or other output to help us see the error

I will provide any and all log files that anyone would like to see.

krisgesling, 2021-05-03:
Hi there, the Picroft image isn't really intended to be a wake word training system and it's not something that we test releases for. 

I'd try a standard install of Precise for training, then copy your resulting model files to your Picroft to use it as your wake word. 

I'll leave this issue open though, others might also be training Precise models on their Picrofts and have some advice to provide.

sparky-vision, 2021-05-03:
Understood. Let me ask this, then, instead, since I've since set up a VM to do modeling. I realize this might count as a new issue, so I can close this and make a new one if you like.

I'm getting excellent results with precise-listen, when training my new model. Mycroft, however, will not activate with the .pb. As I understand (perhaps incorrectly) Mycroft is just calling precise "in the background" so to speak...why would it work with precise-listen, but not in Mycroft?

To be clear, it doesn't activate...at all. It acts like it doesn't hear the wakeword. I know it _can_ hear me, though, because it hears (and accurately transcribes) the words that come after the wakeword, and it hears "Hey Mycroft" perfectly. Are there any troubleshooting steps I can do to find out where the disconnect is? Is my understanding of how Mycroft calls precise incorrect? I've been working with the guys over at the mycroft-precise repo and what we've come up with so far is that I need to train my model quite a lot more, but I wanted to see if perhaps someone here had any ideas to share.

meonkeys, 2022-05-14:
Continued at https://community.mycroft.ai/t/for-those-having-trouble-training-a-new-wake-word/10749 , which leads to https://github.com/sparky-vision/mycroft-precise-tips