MycroftAI/enclosure-picroft #156: Error 0x80004005 (Unspecified error) when extracting.

AwesomeAsh313, 2021-08-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the ZIP file that has the IMG file inside of it.
2. Extract the file.
3. Watch it give you error 0x80004005 (Unspecified error)

**Expected behavior**
I expected it to extract the file, then for me to use Balena Etcher on my Linux system to burn the IMG to my Micro SD card, which I would then put back into the Pi 4 and boot up from.

**Log files**
No log files, didn't even get on the Raspberry Pi 4.

**Environment (please complete the following information):**
 - Device type: Raspberry Pi 4
 - OS: Picroft
 - Mycroft-core version: 20.08
 - Other versions: Unknown

**Additional context**
None.

Please think carefully about whether you have modified anything in Mycroft's code or configuration files.  If so, can you reproduce this on a clean installation of Mycroft? Many "bugs" turn out to be non-standard configuration errors.

prasket, 2021-08-22:
Hello there :wave: I was just able to download, extract, burn the image `Picroft Stable 2020-09-07 image`  (20.08) &  `Picroft Release Candidate 2021-06-04 image` (21.02)  to an 32GB sdcard using balenaEtcher on a Linux client. 

You might try re-downloading the images, checking the sha256sum, and trying again. 

Another alternative would be for you to build your own img to burn to sdcard. https://github.com/MycroftAI/enclosure-picroft/blob/buster/image_recipe.md 

Good Luck!

krisgesling, 2021-08-23:
Thanks for verifying it @prasket 

@AwesomeAsh313 can you post the filename that you downloaded, and as prasket suggested please verify the SHA-256 sums against the repo README here:
https://github.com/MycroftAI/enclosure-picroft#picroft---buster-keaton-pork-pi-release

If you're using balenaEtcher you also don't need to extract the archive first, it can do this for you.

AwesomeAsh313, 2021-09-03:
I found out about balenaEtcher 3 days later, and that worked! Thanks!

closed 2021-09-03 by TheDevPlutonum