MycroftAI/enclosure-picroft #157: Make use of GitHub Releases vs Google Drive for images

prasket, 2021-08-23:
Currently the pi images that are offered pre-built are hosted on Google Drive. 

I propose the project quit using Google Drive for these images and embrace using Github releases for hosting the within the repo. https://github.com/MycroftAI/enclosure-picroft/releases 

Having the image stored in Google goes against the privacy mindset of the project as many know Google is one of the most privacy violating companies in the world.

krisgesling, 2021-08-23:
Hey Prasket, 

There's a size limit on the Github releases that Picroft far exceeds. However you can find all the Picroft releases here: 
https://downloads.mycroft.ai/releases/picroft/

We had a brief outage a little while ago and I switched the download links to GDrive while we were investigating but clearly didn't  point them back. This has been fixed up now.

Thanks for the nudge :)

prasket, 2021-08-23:
Ahh makes sense, Thank you for update @krisgesling