MycroftAI/enclosure-picroft #159: picroft fails to update with AIY hardware

zero-hound, 2021-11-04:
**Describe the bug**
When using Google's AIY voice kit as the mic and speaker for Picroft, following the intial setup and installation apt is unable to complete any updates. This is due to the repository at "https://dl.google.com/aiyprojects/deb stable Release" not being signed. The repo must have the trusted=yes option declared in the config file first.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup current release of Picroft
2. Select Google's AIY hardware as the mic and speaker during setup
3. After setup is completed, run `sudo apt update`
4. To resolve error, edit etc/apt/sources.list.d/aiyprojects.list
5. Change `deb https://dl.google.com/aiyprojects/deb stable main` to `deb [trusted=yes] https://dl.google.com/aiyprojects/deb stable main`

**Expected behavior**
`Sudo apt update` should run without terminating due to error

**Log files**
(.venv) pi@picroft:~ $ sudo apt update --allow-releaseinfo-change
Ign:1 https://dl.google.com/aiyprojects/deb stable InRelease
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Hit:3 http://archive.raspberrypi.org/debian buster InRelease                    
Get:4 https://dl.google.com/aiyprojects/deb stable Release [934 B]
Get:5 https://dl.google.com/aiyprojects/deb stable Release.gpg [819 B]
Ign:5 https://dl.google.com/aiyprojects/deb stable Release.gpg
Reading package lists... Done
W: GPG error: https://dl.google.com/aiyprojects/deb stable Release: The following signatures were invalid: EXPKEYSIG 6494C6D6997C215E Google Inc. (Linux Packages Signing Authority) <linux-packages-keymaster@google.com>
E: The repository 'https://dl.google.com/aiyprojects/deb stable Release' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

**Environment (please complete the following information):**
 - Device type: Raspberry Pi 4
 - OS: Picroft
 - Mycroft-core version: 21.2.1 [master]

krisgesling, 2021-11-04:
Thanks for flagging this.

Not something we'll have time to address immediately but good to have it identified.

denics, 2022-05-07:
Hi all,
fixed by 5408b807a15a9c7b3d3c8d18cdca6daeb5fab9cb it can be closed.

denics, 2022-05-07:
duplicated of #135 fixed with 090cb72ba9a877b170eca02783ea2e54d209df20

closed 2022-05-08 by krisgesling