MycroftAI/enclosure-picroft #160: Cannot download Picroft image

ultracoldatoms, 2022-01-11:
When attempting to download the latest stable Picroft image, I am redirected to a Google Drive page that says:

> Sorry, you can't view or download this file at this time.
> 
> Too many users have viewed or downloaded this file recently. Please try accessing the file again later. If the file you are trying to access is particularly large or is shared with many people, it may take up to 24 hours to be able to view or download the file. If you still can't access a file after 24 hours, contact your domain administrator.

Additionally, when I attempt to navigate to https://downloads.mycroft.ai/releases/picroft/, I get a page saying

> 504 Gateway Time-out

soslug, 2022-01-14:
I also was unable to download the image from the link above however both stable and development releases can be obtained from the Picroft web page below:

https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/get-mycroft/picroft#maintaining-your-picroft

ultracoldatoms, 2022-01-17:
> I also was unable to download the image from the link above however both stable and development releases can be obtained from the Picroft web page below:
> 
> https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/get-mycroft/picroft#maintaining-your-picroft

@soslug Thanks for your comment. Attempting to download Picroft from the page you linked brings up the same Google Drive that gave me trouble before. One can download from it sometimes, but it will often reject downloads for several hours in a day. This should really be fixed.

soslug, 2022-01-17:
> @soslug Thanks for your comment. Attempting to download Picroft from the page you linked brings up the same Google Drive that gave me trouble before. One can download from it sometimes, but it will [...]

I would have to say as an end user my experience was a little different to yours by the sounds of it. Whilst both links may point you to the google download page. The first link - 

https://downloads.mycroft.ai/releases/picroft

Gave me a 504 error directly initially, now when I try I get no error message or download page within a reasonable time frame for this page to be loaded.

https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/get-mycroft/picroft#maintaining-your-picroft a link I got from the Picroft website. eg.

https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/get-mycroft/picroft

This one displays 

Google Drive can't scan this file for viruses.

Picroft_v20.08_2020-09-07.img.zip (3.0G) is too large for Google to scan for viruses. 

Would you still like to download this file?
Download anyway

This was the only one that worked for me, I would point out I am using a VPN to a local node London perhaps that's why I can download and you cannot, I am not sure. Turning off my VPN proves inconclusive as I live near that location anyway, and could access this page for download.

g3holliday, 2022-02-07:
It works for me... it did this a few days ago, but not since

krisgesling, 2022-02-09:
Hey, we were trying a new host for downloads.mycroft.ai and been having some issues which is why we dropped back to Google Drive but it has it's own limits that we're clearly hitting.

I've just setup a new CDN. I want to test it a bit before we go all in but if you want the latest Picroft image you can grab them here and it should be a blazing fast download:
- Picroft (Stable image) - 20.08
  https://mycroft-cdn2.fra1.cdn.digitaloceanspaces.com/Picroft_v20.08_2020-09-07.img.zip
- Picroft (Release Candidate) - 21.02
  https://mycroft-cdn2.fra1.cdn.digitaloceanspaces.com/Picroft_v21.02.0_20210604.img.zip

Let me know how it goes.

g3holliday, 2022-02-09:
Nice!! That is faster....