MycroftAI/enclosure-picroft #162: Speakers not playing

g3holliday, 2022-02-07:
When I complete the guided setup, the speakers don't play at all. I'm pretty sure the microphone works, since I see the input.

To reproduce, select guided setup, then connect network, and complete the setup. On the speaker step, select either 3,5mm or USB - I haven't tested anything else. When you test the audio, it doesn't play. And no, it's not my headphones, I tested at least 3 different pairs.

I expected that the speakers should output sound feedback.

**Environment (please complete the following information):**
 - Device type:  Raspberry Pi 4
 - OS: Picroft
 - Mycroft-core version: Picroft Stable 2020-09-07 

I never edited anything in the code...

archonic, 2022-03-27:
I've found that on both Buster and Bullseye, the audio output for 3.5mm jack and HDMI is reversed. If you select HDMI it will output 3.5mm and vice versa. I'm sure this is confusing a lot of people and causing a lot of drop-off for engagement with picroft.

g3holliday, 2022-03-29:
Yeah, I'll try it. Thanks!